gnuradio/gnuradio #5733: Document performance inconsistencies of FFT Filter block

Zlika, 2022-03-31:
### Feature Description

The performance (throughput) of the FFT Filter block is neither constant nor linear with respect to the number of taps (cf. figure below).
Looking at the code, it seems that the performance of this block is directly related to the parameter fft_filter_ccc::d_nsamples (it's the number of input samples processed at the same time), whose value if d_nsamples = d_fftsize - d_ntaps + 1. d_fftsize also depends on the number of taps.
In the following figure, we can see that it's possible to have an important performance boost by carefully choosing the number of taps of the filter, i.e. by 0-padding a filter to reach the next performance threshold.
This behavior is not intuitive and then should be documented.
[image]

### Feature Urgency

low (just an idea)

Zlika, 2022-03-31:
Ok. I've just tried to create an account to be able to edit the wiki, by the confirmation link is not working ("Invalid confirmation code. The code may have expired.") :-(

marcusmueller, 2022-03-31:
Thank you, and: oh! OK, I'm letting the website experts know.

drmpeg, 2022-04-01:
An old article relating to this.

http://www.trondeau.com/blog/2014/2/27/to-use-or-not-to-use-fft-filters.html