gnuradio/gnuradio #5734: M1 macOS: UI becomes unresponsive after connectign two nodes

nils-werner, 2022-03-31:
### What happened?

After trying to connect two nodes the UI becomes unresposive.

### System Information

OS: MacOS 12.1 Monterey
Architecture: M1, x86 Emulation on
GR Installation Method: `arch -x86_64 /usr/local/bin/brew install gnuradio`

### GNU Radio Version

3.9  (maint-3.9)

### Specific Version

3.9.3.0

### Steps to Reproduce the Problem

1. Obtain M1 mac
2. Install Homebrew
3. Install GNURadio using `arch -x86_64 /usr/local/bin/brew install gnuradio`
4. Run `gnuradio-companion`
5. Create two nodes
6. Connect them
7. The UI will flicker and will become unresposive. Buttons can still be pressed but the interaction is not represented visually.

### Relevant log output

```shell
Nothing, despite running `gnuradio-companion --log debug`
```

nils-werner, 2022-03-31:
I have also discovered another aspect of this: It seems that while dragging a connection the GUI is rapidly toggling between the connection and the most recently selected node. The node box flashes quickly blue and black and the connection appears and disappears.

Etherealflux, 2022-04-03:
Loading a .grc file that contains a connection between two nodes causes the window to flicker rapidly. This stops after about a second; after that point, it's unresponsive.

Niggi22, 2022-04-04:
Have the same Problem here. Sadly it's almost unusable.

elimbaum, 2022-10-11:
This is still an issue and not limited to M1 macs.

Install info:
- brew, version 3.10.4.0 (python 3.10.7)
- macOS Monterey 12.6 (Intel)

Exactly same behavior as observed by @nils-werner:

1. Open an empty flowgraph.
2. Add two blocks. (No issues)
3. Connect the blocks.
4. UI flickers and then freezes.

UI is still responsive to input; i.e. right click menu appears, though the flowgraph doesn't change. Interestingly, if I delete one of the blocks from the right click menu, the UI responds again.

While trying to grab some screenshots to demonstrate this behavior, I realized that this is _also dependent on the size of the companion window!_ If the GUI freezes or goes black, I can always recover by shrinking the window.

Messing around with it a bit more, it seems like the actual quantity triggering the bug is the size of the _main_ flowgraph area (not counting the console/variable/blocks areas). I can't quite figure out the threshold – it seems to be jumping around – but I can also recover from a frozen GUI by grabbing the console/variable resize bar and shrinking the size of the main flowgraph area.

I'd love to help get this resolved as this is making GRC basically unusable on Mac.

dkozel, 2022-10-13:
@elimbaum, Any chance that size is approx 1/4th the screen width/height? There's a bug somewhere in grc which causes scaling issues which present as the flowgraph area only filling 1/4th of the correct area.

ryanvolz, 2022-11-10:
Any chance #6352 fixes this? That's a python-only fix, so you can manually edit the appropriate installed python files to test it.

jimilinuxguy, 2022-11-17:
> Any chance #6352 fixes this? That's a python-only fix, so you can manually edit the appropriate installed python files to test it.

Confirmed that this worked for me. Thanks!

closed 2022-11-17 by mormj

elimbaum, 2022-12-01:
Hadn't gotten a chance to test this for a few weeks. Just wanted to confirm I manually applied the changes from #6352 and this **does** work! I can no longer reproduce the above bug, either by changing window size or by moving between monitors. Thanks!