gnuradio/gnuradio #5736: Imported target "pybind11::module" includes non-existent path

AdolfsGrandson, 2022-04-01:
```
# cmake -DCMAKE_BUILD_TYPE=Release -DPYTHON_EXECUTABLE=/usr/bin/python3 ../
-- Build type set to Release.
-- Build date set to Fri, 01 Apr 2022 16:20:47Z.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- Extracting version information from git describe...
-- 
-- Configuring testing-support support...
--   Dependency Boost_UNIT_TEST_FRAMEWORK_FOUND = 1
--   Enabling testing-support support.
--   Override with -DENABLE_TESTING=ON/OFF
-- Compiler Version: cc (GCC) 11.2.0
Copyright (C) 2021 Free Software Foundation, Inc.
Это свободно распространяемое программное обеспечение. Условия копирования
приведены в исходных текстах.

  Без гарантии каких-либо качеств, включая 
коммерческую ценность и применимость для каких-либо целей.
-- Compiler Flags: /sbin/cc:::-O3 -DNDEBUG -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/sbin/c++:::-O3 -DNDEBUG -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection -Wp,-D_GLIBCXX_ASSERTIONS -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.10.4", minimum required is "3.6.5") 
-- Python checking for packaging - found
-- Python checking for numpy - found
-- Python checking for pygccxml - not found
-- 
-- ***************************** WARNING!!! *************************
-- pygccxml is highly recommended for using gr_modtool
-- and is either not present or below the minimum version 2.0.0
-- Only trivial bindings will be generated using gr_modtool bind 
-- ******************************************************************
-- Found pybind11: //include (found version "2.8.1" )
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency PACKAGING_FOUND = TRUE
--   Dependency pybind11_FOUND = 1
--   Dependency NUMPY_FOUND = TRUE
--   Enabling python-support support.
--   Override with -DENABLE_PYTHON=ON/OFF
-- 
-- Configuring VOLK support...
--   Found VOLK:
--   * Version: 2.5.0
--   * Libraries: Volk::volk
--   * Includes: /usr/include
-- Not using additional GNU Radio native architecture optimizations.
-- Using GMP.
-- 
-- Configuring post-install support...
--   Enabling post-install support.
--   Override with -DENABLE_POSTINSTALL=ON/OFF
-- 
-- Configuring doxygen support...
--   Dependency DOXYGEN_FOUND = YES
--   Enabling doxygen support.
--   Override with -DENABLE_DOXYGEN=ON/OFF
-- 
-- Configuring man-pages support...
--   Enabling man-pages support.
--   Override with -DENABLE_MANPAGES=ON/OFF
-- Using GMP.
-- 
-- Configuring gnuradio-runtime support...
--   Dependency Boost_FOUND = TRUE
--   Dependency PYTHONINTERP_FOUND = TRUE
--   Dependency MPLIB_FOUND = TRUE
[... 601 lines cut ...]
  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Error in gr-soapy/python/soapy/bindings/CMakeLists.txt:
  Imported target "pybind11::module" includes non-existent path

    "/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

```

AdolfsGrandson, 2022-04-01:
UPD: ununstalling arch package
`sudo pacman -Rdd pybind11`
and installing python package
`pip install "pybind11[global]"`
solved this problem

AdolfsGrandson, 2022-04-01:
UPD: uninstalling arch package
`sudo pacman -Rdd pybind11`
and installing python package
`pip install "pybind11[global]"`
solved this problem

ryanvolz, 2022-04-02:
It seems like the CMake config file provided by the arch package of pybind11 is inserting the broken include directory. GNU Radio doesn't do anything special there and relies on the `pybind11::module` target that the package provides. Maybe check with the arch pybind11 packagers?

marcusmueller, 2022-04-03:
It seems this was a broken package and got fixed by reinstallation; closing, but please reopen if the problem persists.

closed 2022-04-03 by marcusmueller