gnuradio/gnuradio #5742: PIP-installable block libraries

nils-werner, 2022-04-04:
### Feature Description

It would be great if pure Python blocks could not only be installed using pyBOMBS but also using PIP.

Python packages have the ability to [advertise their behaviour](https://setuptools.pypa.io/en/latest/userguide/entry_point.html#dynamic-discovery-of-services-and-plugins), e.g. so that plugins can become automatically discoverable.

This means that a Python package `mypackage` could contain GNURadio blocks (among other things) and would advertise itself in its `setup.py` metadata with

```python
setup(
    # ...
    entry_points={
        "gr.blocks": [
            "hello-world = mypackage:hello_world",
        ],
    },
)
```

GNURadio can then discover all of these modules using

```python
from importlib import metadata
eps = metadata.entry_points()['gr.blocks']
for ep in eps:
    block = ep.load()
```

And show them in the GUI.

### Feature Urgency

medium (would be nice to have in the near future)

willcode, 2022-04-04:
Is there anything special about Python block, with respect to pip/wheel/etc? The install methods should be consistent for the system, so we don't end up with a Python block ecosystem and a general OOT ecosystem.

nils-werner, 2022-04-04:
> Is there anything special about Python block, with respect to pip/wheel/etc? 

I am not sure I understand.

> The install methods should be consistent for the system, so we don't end up with a Python block ecosystem and a general OOT ecosystem.

I am trying to combine the SciPy and Machine Learning ecosystem with GNURadio.

SciPy and ML are usually installed and maintained by Conda and PIP while GNURadio comes as a OS package (and more recently as a Conda package) and pyBOMBS OOT modules. There is simply a large disconnect between these two ecosystems that I would like be able to bridge.

Coming from the Python side, using pyBOMBS is very unnatural for me. Right now GNURadio requires me to use pyBOMBS to compile and install blocks, even if it is a pure Python module and there is nothing really to compile. And I am not sure how I would have to handle Python dependencies in my block, and if pyBOMBS could handle them.

And I think that relying on pyBOMBS alone is unwise because you'll be locking out most of the SciPy ecosystem.

As a first step I think that if GNURadio was able to discover Python-only blocks that were installed "the Python way" it would be a first step to close the disconnect.

Also, if GNURadio and all OOT modules were installed by Conda (and not by an OS package and pyBOMBS), GNURadio would become a first-class-citizen in the SciPy/ML ecosystem and could immediately benefit from all the libraries that are out there. But that's for another ticket.

marcusmueller, 2022-04-04:
Re:
> Is there anything special about Python block, with respect to pip/wheel/etc?

I think what @willcode says here is: in what way do we *want* python-only modules to differ from C++-containing ones? (and if I may put a bit of an undertone to that: I don't think we want that!)

Your ML ecosystem also contains a lot of compiled code! So, as @willcode says, when we start doing this, we should, for consistency, probably go all-in and offer compiled/compiling pypi packages as well (as, say tensorflow and numpy). As your impression that we're relying on pyboms shows, confusion about how to get GNU Radio and OOT components is one significant user show stopper!

> Right now GNURadio requires me to use pyBOMBS to compile and install blocks, even if it is a pure Python module and there is nothing really to compile. 

Let us shortly address pybombs: we're really not relying on it; it's been falling out of favor a bit lately and is definitely not the default installation method we generally recommend. Yes, that means that currently the default way of installing GNU Radio out-of-tree modules is by manually building them from source, and we'd like to change that, I think.

But, honestly, using >= 4 lines of pybombs is not better than `git clone; mkdir/cd build; cmake -DCMAKE_INSTALL_PREFIX=$VIRTUAL_ENV ..; cmake --build . ; cmake --install .`, and that should fully suffice for you, I guess, especially for py-only modules. (I'm not saying this is a good situation, I'm just hopefully making it easier for you to get around without pybombs within your virtualenv/conda based workflow. I don't think we've officially said so, but let's consider pybombs a tool that's deprecated for this kind of use case.)

Now, conda is a nice thing, because we are already invested in conda packaging of GNU Radio – mostly in the person of @ryanvolz , whose days sadly only have 24hrs :)

So, yeah, what I think would be an interesting thing to look at is including a setup.cfg and maybe a pyproject.toml in the OOT template. It's certainly been on my to-look-at-one-day list for quite a while, but never gotten around to, so that's why I like your proposal!

nils-werner, 2022-04-05:
Thanks for your thorough and insightful responses!

> But, honestly, using >= 4 lines of pybombs is not better than `git clone; mkdir/cd build; cmake -DCMAKE_INSTALL_PREFIX=$VIRTUAL_ENV ..; cmake --build . ; cmake --install .`

There is a big difference if the command is a package manger, because it can also install dependencies which cmake cannot do. So a single line `pip install` or `conda install` actually does a lot more than just run `make`.

This also leads to reproducible environments where you can manage, snapshot and recreate environments reliably. Replacing a `cmake` with `pip install` does indeed not sound like much, but exactly recreating an entire environment (or somebody else's environment) on a different machine is really difficult if I have to do every step by hand.

This is where true package managers become irreplaceable.

> Returning again to the actual issue here, I think there is a decent argument for being able to add to the GRC blocks path by way of registering an entry point like @nils-werner describes. Basically making the manual step of editing the GRC config file as described [here](https://github.com/MITHaystack/digital_rf#installation) unnecessary.

Exactly.

Another benefit: PIP allows for "editable installs". The source files of these installs never leave their original location, but merely their path and entry_points are registered in the environment. That means I can Git clone a library that I want to work on, editable-install it and start working on the sources in their original location. My Python environment will pick them up regardless. The way I envision this feature is that GNURadio would behave similarly.

> I'm of the opinion that PyPI really only works well for Python-only things, and the conda ecosystem is the better hope for achieving those goals.

Agreed, PyPI is not the way to go. Conda is much more suitable for this. I am only suggesting something like the entry_point hook here, not to adopt PIP/PyPI as the packaging format.

It would be really awesome if I could write a DSP lib in Python and it can, by just adding a couple of lines of boilerplate, also ship a GNURadio block that also registers itself into the appropriate places.

One thing that is really unfortunate about Conda is the fact that it cannot install packages from source. It would be really awesome if you could just say

```
conda install git+https://github.com/MITHaystack/digital_rf
```

but I don't think that's possible. If it were, there wouldn't really be a need for a bioconda-like feed but people could just pull their blocks from GitHub etc.

A possible alternative towards self-registering blocks would be to change the `grc.conf` format so that `local_blocks_path` can be extended by config file fragments, but that's just a thought.

willcode, 2022-04-05:
A related consideration - let's make sure we don't end up with solutions where "too much" arbitrary code gets installed on a user machine without a packager (human) reviewing somewhere along the line. Right, PyBOMBS doesn't help here. Some sort of pre-packaged format at least _implies_ that someone has looked at a version of the package. Don't want us to get in the situation where things break often (or worse) due to changes in someone's repo.

nils-werner, 2022-04-13:
Actually it turns out can be achieved by the Python package shipping the `.block.yaml` file, and dropping it in the right place using `data_files`:

```python
setuptools.setup(
    # ...
    data_files=[(
        "share/gnuradio/grc/blocks",
        ["grc/my.block.yml",],
    )]
)
```

closed 2022-04-19 by nils-werner