gnuradio/gnuradio #5756: Unuseful QT GUI Chooser

zhanght34, 2022-04-08:
### What happened?

    I create oot block named 'Link16MsgV2',use it to receive the value defined by 'QT GUI chooser' and then output the corresponding PDU.
    But ,when I run my grc and change the choose of ‘QT GUI chooser’ ,it is invalid. In other words, when I run grc, the value of the modified ‘QT GUI chooser’ doesn’t pass to my oot block.The value of ‘QT GUI chooser’ is always default optin.
    Does anyone know the problem with this? Thanks very much!!!
    This is my first time asking a question. If you feel that there is something I didn't express clearly, I will add an explanation.
[image]
[image]
[image]
[image]

### System Information

OS: (Linux Ubuntu 20.04)
GR Installation Method: (follow the official website of  GNU Radio)

### GNU Radio Version

3.11-git (main)

### Specific Version

3.11.0.0

### Steps to Reproduce the Problem

1. create grc like this
[image]
2.'Link16MsgV2' BLOCK like this
[image]
3.run this grc
[image]
4.when i chang the chooser of timeslot,the printf debuge is no changed
[image]

willcode, 2022-04-08:
@zhanght34 Thanks for your question. We use the issue tracker for bug reports and feature requests. For help, please use chat.gnuradio.org or the mailing list.

closed 2022-04-08 by willcode

zhanght34, 2022-04-08:
> please use chat.gnuradio.org or the mailing list.

okok,understand，thanks.