gnuradio/gnuradio #5757: how to fix that？

xhtechxposed, 2022-04-08:
### Feature Description

error: RPC failed; curl 56 GnuTLS recv error (-9): Error decoding the received TLS packet.
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
error: Could not fetch uhd

### Feature Urgency

high (impacts broad portion of ecosystem)

marcusmueller, 2022-04-08:
This is not a feature request, but you explicitly used  the "Feature Request" issue template. Also, your title and your question do not describe at all what you're doing, and when this happens. So, closing in its current form. Please use the "Bug Report" issue template if you're sure this is a bug that relates to GNU Radio (doesn't really look like it).

closed 2022-04-08 by marcusmueller

mormj, 2022-04-08:
Also, if you're not sure it's a bug but just need technical help getting something gnu radio related working, please use the mailing list (https://wiki.gnuradio.org/index.php?title=MailingLists) or the chat on chat.gnuradio.org