gnuradio/gnuradio #5759: update rate in QT GUI Frequency Sink is unintuitive/could be improved

muaddib1984, 2022-04-10:
### What happened?

Running a flowgraph with QT GUI Frequency sink and an update rate of 0.01 at sample rate of 48k with various FFT sizes yields sluggish/jerky response (the opposite of the expected behavior). If the FFT size is changed from the initial value, then changed back, on some systems the behavior is corrected, on others it's not.
After discussion with @argilo on the dev channel it was found that "the current work function implementation is basically 'if at least <fft size> samples have arrived and at least <update rate> time has passed, then update the display'."
This could be improved by a change in implementation similar to GQRX with a timer and sample buffer.

### System Information

GNURadio v3.10.2.0-rc2
Ubuntu 20.04 and 21.04
multiple hardware platforms/core counts

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.2.0-rc1

### Steps to Reproduce the Problem

run a flowgraph and set the QT Freq GUI with update rate of 0.01, run at high/low sample rates (2Msps/48ksps)

marcusmueller, 2022-04-10:
I like this. But I'll carefully point out that this might be indicative of the whole qtgui approach (which looks the same as this, also for waterfall sink, which is even more confusing) being in need of an overhaul.

argilo, 2022-04-10:
Gqrx solves this problem like so:

* The `work` function just copies samples into a circular buffer. The buffer is implemented using GNU radio's double-mapped buffer, and is long enough to accommodate approximately 1 second of samples.
* A `QTimer` drives GUI updates. Each time the timer fires, the timer handler checks how much time has passed since the last invocation and deletes a corresponding amount of samples from the circular buffer (plus a fudge factor, to ensure that a backlog doesn't build up due to the two-clock problem, and taking care to leave enough samples for one FFT). Then it uses the oldest remaining samples in the buffer to draw the FFT.

This ensures that FFT updates are smooth, even if the samples arrive in a bursty fashion (which they often do, due to buffering in SDR hardware).

argilo, 2022-04-10:
One disadvantage to doing it this way in GNU Radio would be that if the user set the Frequency Sink's sample rate parameter too low, then the circular buffer would fill up and the display would be delayed. Gqrx doesn't have this problem because it knows what the actual sample rate is.

argilo, 2022-04-10:
Unthrottled flow graphs would also have unusual behaviour (with lots of input samples skipped), but that's already the case with the current implementation. Perhaps there should be an option to use the incoming samples as a time reference instead of the wall clock.

khimaros, 2022-11-23:
is there any way to work around this as a user?

i've tried various settings for bandwidth, FFT size, and update period and can't seem to produce something with a decent refresh rate.

argilo, 2022-11-24:
Unfortunately I doubt there is a reliable workaround, since the QT GUI blocks are currently at the mercy of the scheduler. You could maybe try reducing output buffer size, as described in the "Latency and Throughput" section here: 

https://wiki.gnuradio.org/index.php/Handling_Flowgraphs

Reducing buffer size may force the scheduler to call the QT GUI block's `work` function more often.

argilo, 2023-08-15:
> If the FFT size is changed from the initial value, then changed back, on some systems the behavior is corrected

This behaviour, as well as excessively slow refresh rates, were an inadvertent side effect of #5442, which configured the block to consume 32768 samples at a time upon startup, regardless of the actual FFT size. I've fixed that in #6788, which should drastically improve the refresh rate when FFT sizes smaller than 32768 are used.

n7ihq, 2023-12-09:
> is there any way to work around this as a user?
> 
> i've tried various settings for bandwidth, FFT size, and update period and can't seem to produce something with a decent refresh rate.

The QT GUI Sink with Plot Frequency ON works well with a steady refresh rate at all FFT size settings.

jephthai, 2025-03-18:
> > is there any way to work around this as a user?
> > i've tried various settings for bandwidth, FFT size, and update period and can't seem to produce something with a decent refresh rate.
> 
> The QT GUI Sink with Plot Frequency ON works well with a steady refresh rate at all FFT size settings.

Wow, thank you for posting this here.  I've been flipping the FFT size twice every time I started a sketch for... geez... years, I guess.  It really adds up over time, and now I'll just use the QT GUI Sink and my grumbling quotient will go way down.

argilo, 2025-03-18:
The bug was fixed in GNU Radio 3.10.8.0 (released October 2023), so upgrading will also resolve the problem.

PR #6788 (2023-08-16): Set the frequency sink's output multiple on start
## Description
The refresh rate of the QT GUI Time Sink can be painfully slow when the input sample rate is low. This was a side effect of #5442, which set the block's output multiple to `MAX_FFT_SIZE` = 32768 in the block's constructor, presumably to ensure that the scheduler allocates a large enough buffer to accommodate FFT size changes made in the GUI at runtime:

https://github.com/gnuradio/gnuradio/blob/0966207dbc5cc0077f7ee6cde0aaa95f7c7a8d99/gr-qtgui/lib/freq_sink_c_impl.cc#L128

If the FFT size is changed at runtime, the refresh rate s [... cut]