gnuradio/gnuradio #5762: Small block size with tailbiting CC results in errors with no noise

jaredd, 2022-04-11:
### What happened?

The CC encoder / decoder in tailbiting mode in particular have a minimum frame size due to the tail length defined on https://github.com/gnuradio/gnuradio/blob/9e6aeefeae8294026a188d5597c4f7559179dc5c/gr-fec/lib/cc_decoder_impl.cc#L77 and utilized in https://github.com/gnuradio/gnuradio/blob/9e6aeefeae8294026a188d5597c4f7559179dc5c/gr-fec/lib/cc_decoder_impl.cc#L339.

Based on the parameters it appears the minimum block size must be strictly > 4 * (K - 1) bits where K is the contraint length.

1. Using the [gr-fec/examples/fecapi_cc_decoders.grc](https://github.com/gnuradio/gnuradio/blob/v3.10.0.0/gr-fec/examples/fecapi_cc_decoders.grc) example, set the CC mode to TAILBITING.
2. Set the frame_size to 4 (bytes) (note: threshold 4 * (K - 1) = 24 bits).
3. Run the flowgraph and observe no errors in the decoded stream.
4. Set the frame_size to 3 (bytes).
5. Run the flowgraph and observe errors in the decoded stream.

I would expect either an error or warning that the frame size was too small for the implementation and configuration. Otherwise, the user is left scratching their head why they're getting errors with 0 noise.

### System Information

OS: Debian Bullseye (Linux 198311bceba5 5.15.23-76051523-generic #202202110435~1644952300~21.10~96763f1 SMP Tue Feb 15 19:52:40 U x86_64 GNU/Linux)
GR Installation Method: Source installation

GR running in a Docker container.

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.0.0

### Steps to Reproduce the Problem

1. Using the [gr-fec/examples/fecapi_cc_decoders.grc](https://github.com/gnuradio/gnuradio/blob/v3.10.0.0/gr-fec/examples/fecapi_cc_decoders.grc) example, set the CC mode to TAILBITING.
2. Set the frame_size to 4 (bytes) (note: threshold 4 * (K - 1) = 24 bits).
3. Run the flowgraph and observe no errors in the decoded stream.
4. Set the frame_size to 3 (bytes).
5. Run the flowgraph and observe errors in the decoded stream.

marcusmueller, 2024-01-31:
yep, far as I can tell from the source, still there.

marcusmueller, 2024-03-11:
would be high, but feeling is to fix a hoist of things with gr-fec, this would be rewritten inherently