gnuradio/gnuradio #5766: gr_modtool newmod should include .gitignore

nils-werner, 2022-04-13:
### Feature Description

It would be really handy if the skeleton created by `gr_modtool newmod` included a best practices `.gitignore` file.

Right now I have to puzzle one together from several templates out there: [cmake](https://github.com/github/gitignore/blob/main/CMake.gitignore), [python](https://github.com/github/gitignore/blob/main/Python.gitignore), [C++](https://github.com/github/gitignore/blob/main/C++.gitignore) and I am sure I am missing something.

### Feature Urgency

medium (would be nice to have in the near future)

### More Information

I'd be willing to submit a PR, I just haven't found the right location in the source tree.

drmpeg, 2022-04-13:
I think it's a bug. There used to be a ```.gitignore``` (it's still there in maint-3.9), but it was mistakenly(?) moved in commit https://github.com/gnuradio/gnuradio/commit/d3e49a4e0c72ab81d18bf479d2f6cdccf443b616 to the python directory.

The contents is:
```
*~
*.pyc
*.pyo
build*/
examples/grc/*.py
```

@mormj

mormj, 2022-04-13:
Yep, that one's on me.  It should be there but looks like I mixed it up in the shuffle that happened in v3.10.  The old one should be sufficient, but if you see other filetypes that we should have, feel free to expand it.