gnuradio/gnuradio #5777: uuuuuuuuunderflows and cmd time errors

zhanght34, 2022-04-21:
### What happened?

I'm playing with gnuradio and an USRP X410 device.

The issue:
1.fpga image:x4_200     Even the simplest flow graph generated by grc and containing UHD USRP Sink gives lots of 'U's if the sampling rate is larger than about 100MHz. At same time if i set four channel in USRP Sink. GNU Radio will prints "cmd time errors"

2.fpga image:CG_400      When I run the GRC flow graph, the sample rate and TX rate of the USRP are automatically set to 491.52MHz, and then "uuuuuuuuunderflows" is printed, is this a bug? How to fix it?

### System Information

Ubuntu20.04
GNU Radio 3.10.1.1
Python  3.8.10
UHD 4.2.00-rc1

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.1.1

### Steps to Reproduce the Problem

[image]
[image]

marcusmueller, 2022-04-21:
1. Not an issue – your computer simply isn't fast enough at supplying sample packets at this rate, it seems.
2. a hardware restriction of your USRP – it simply doesn't support 5 MHz as a sampling rate with the master clock rate used. Try setting the master clock rate to `500e6` (you're using the [CG_400 image](https://files.ettus.com/manual/page_usrp_x4xx.html#x4xx_updating_fpga_types)). You can only use integer factors of your master clock rate as sampling rate.

closed 2022-04-21 by marcusmueller

zhanght34, 2022-04-21:
My computer printfs underflow after setting the sampling rate above 100MHz,.But the computer CPU is not fully loaded. The connection between the computer and the USRP X410 is through QFSP. Could it be a problem with the TCP buffer settings?
And do you know 'cmd time errors'?
Thanks very much!!!I've been bothered by it for two  weeks.

LittleHong12138, 2023-03-31:
I also met the same problem. Have you solved it？