gnuradio/gnuradio #5778: cmd time errors with UHD USRP Sink

zhanght34, 2022-04-22:
### What happened?

I'm playing with gnuradio and an USRP X410 device.
I connect pc with USRP X410 by QFSP.I want to test how fast the QFSP port transfer rate is.
Then in gnuradio grc,i have connected two block:Sig Source and USRP Sink. Details as shown
[image]
：
When i run this grc,and set samp_ate to 30MHz，gnuradio printfs"cmd time errors".How to slove it?
[image]

### System Information

i9-11900F @ 2.50GHz
Ubuntu20.04
GNU Radio 3.10.1.1
Python 3.8.10
UHD 4.2.00-rc1

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.1.1

### Steps to Reproduce the Problem

1.Run this grc
2.click ‘samp_rate’ up button to 30MHz.

Then,gnuradio printfs"
                 Susrp_sink :error: In the last 28292 ms, 1 cmd time errors occurred.
                 SSSSSSSSUusrp_sink :error: In the last 28298 ms, 1 underflows occurred.
"
At the same time, the light on the usrp x410 showing the RF transmit go off.

### Relevant log output

```shell
Susrp_sink :error: In the last 28292 ms, 1 cmd time errors occurred.
SSSSSSSSUusrp_sink :error: In the last 28298 ms, 1 underflows occurred.
```

marcusmueller, 2022-04-22:
These are not GNU Radio bugs, please don't report them here! The usrp-users mailing list might be a better place :) 

Also I pointed you to setting the `master_clock_rate` before, it seems that hasn't worked!

closed 2022-04-22 by marcusmueller

zhanght34, 2022-04-23:
Thanks ,help me very very timely.Meanwhile, I'm sorry for my wrong approach.