gnuradio/gnuradio #5779: Adding node sets `state: true` disabling and enabling it sets `state: enabled`

nils-werner, 2022-04-22:
### What happened?

When I add a node to my flowgraph, the `.grc` file will contain a line

```
state: true
```

If I disable that node, the line switches to

```
state: disabled
```

and if I re-enable the node, it switches to

```
state: enabled
```

I think the value `true` is incorrect and it should be `enabled` instead.

### System Information

OS: Arch Linux
GR Installation Method: Conda

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.1.1

### Steps to Reproduce the Problem

1. Add a node. The file says `state: true`
2. Disable it. The file says `state:disabled`
3. Enable it. The file says `state:enabled`

haakov, 2022-04-23:
You're right! This is the culprit:
https://github.com/gnuradio/gnuradio/blob/c6d094d79e420c34f7726b893691fdb43eab7360/grc/core/blocks/block.py#L79-L80

Sthyarala, 2022-09-21:
@nils-werner I will be happy to take this issue.

dkozel, 2022-09-29:
@Sthyarala If you would like to fix this problem and open a pull request that would be great, please do!

PR #6609 (2023-04-05): grc: make node states consistant

just a one line change to make sure node state is consistant
## Description

## Related Issue

Fixes #5779

## Which blocks/areas does this affect?

## Testing Done

ma
closed 2023-04-05 by willcode