gnuradio/gnuradio #5780: IIO PlutoSDR Sink block output is very low

computer-wizard, 2022-04-24:
### What happened?

Maximum output caused by the IIO PlutoSDR Sink block is very low. I did a comparison with the Soapy Pluto Sink block and both set for max output of the Pluto and the output of the IIO PlutoSDR Sink block + Pluto is about 10 db lower as by the Soapy Pluto Sink block.  I used the same layout and just replaced the IIO Pluto block with the Soapy version.
Output of the  IIO PlutoSDR Sink block is also very low compared with the time in the past when gr-iio was still an OOT module.

### System Information

OS: Ubuntu 20.04  GnuRadio 3.10.1.1  was just installed from a package from the gnuradio ppa repository

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.1.1

### Steps to Reproduce the Problem

Compare IIO PlutoSDR Sink block with Soapy PlutoSDR Sink block  Just use the same layout and replace IIO PlutoSDR Sink
block with the Soapy version. In my case I used a ZMQ REQ Source block connected to the IIO PlutoSDR Sink or the Soapy Pluto Sink. The feed was a ODR tools DAB signal.

### Relevant log output

```shell
Not relevant
```

adisuciu, 2022-04-28:
I just sent a PR that fixes setting the PlutoSDR sink attenuation https://github.com/gnuradio/gnuradio/pull/5788 . I think this patch may fix the problem ..

computer-wizard, 2022-04-28:
That should be most welcome, thanks!

computer-wizard, 2022-06-07:
GNU Radio version  3.10.2.0-0 (deb package) was released this week and I installed it. I can report that the low output of the IIO
PlutoSDR sink block is still there.

mormj, 2022-08-19:
This fix would not have made it into 3.10.2.0, but should be in 3.10.3.0

closed 2022-08-19 by mormj