gnuradio/gnuradio #5781: Cannot use variable binding in message handlers

nils-werner, 2022-04-26:
### What happened?

I am trying to write a Python block with a variable number of message inputs. Of course I don't want to implement a fixed number of message handlers for a variable number of ports, so I would write handler and assign all ports to it, with another parameter indicating which port was used:

```python
class example(gr.basic_block):
    def __init__(
        self,
        receivers=1,
    ):
        super().__init__(
            name="Example",
            in_sig=None,
            out_sig=None,
        )

        for i in range(receivers + 1):
            self.message_port_register_in(pmt.intern(f"receiver{i}"))
            self.set_msg_handler(pmt.intern(f"receiver{i}"), functools.partialmethod(self.receiver_handler(i)))

    def receiver_handler(self, port, message):
        print(port, message)
```

and

```yaml
id: message_example
label: Multi Message Example

templates:
  imports: |-
    import wasn
  make: >
      example(
        receivers=${receivers},
      )

parameters:
-   id: receivers
    label: Receivers
    dtype: int
    default: '2'
    hide: 'part'

inputs:
-   domain: message
    label: receiver
    optional: true
    multiplicity: ${ receivers }

file_format: 1
```

However this is not possible because internally GNURadio is keeping a string reference to my method, instead of a pointer:

 - [gateway.py](https://github.com/gnuradio/gnuradio/blob/main/gnuradio-runtime/python/gnuradio/gr/gateway.py#L228)
 - [block_gateway_impl.h saving](https://github.com/gnuradio/gnuradio/blob/b2c9623cbd548bd86250759007b80b61bd4a2a06/gnuradio-runtime/lib/block_gateway_impl.h#L41)
 - [block_gateway_impl.h calling](https://github.com/gnuradio/gnuradio/blob/b2c9623cbd548bd86250759007b80b61bd4a2a06/gnuradio-runtime/lib/block_gateway_impl.h#L72)

I believe a more portable solution would be to store method pointers instead of strings here.

### System Information

OS: Arch Linux
GR Installation Method: Conda

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.1.1

### Steps to Reproduce the Problem

1. Create block from above
2. Connect message inputs to some message source
3. Generate and run graph

marcusmueller, 2024-01-31:
Hm, saving the function object's address isn't very pythonic, so I'd say that would probably lead to a lot of confusion.

The pythonic way to write your code would be to use python lambdas, and then their lookup would be name-based, too.

closed 2024-01-31 by marcusmueller

nils-werner, 2024-02-01:
I didn't mean the function address but it's reference. So

```
self.set_msg_handler("receiver", self.receiver_handler)
```

instead of 

```
self.set_msg_handler("receiver", "receiver_handler")
```

Could you give an example how this can be achieved using lambdas?

marcusmueller, 2024-02-01:
aahh, now I understand!