gnuradio/gnuradio #5797: Bernoulli Source

michael-roe, 2022-04-30:
### Feature Description

A Bernoulli Source would be a block that outputs a 1 with probability p and a 0 with probability 1-p.

I have an example implementation in
https://github.com/michael-roe/gr-c4fm

This is useful, for example, in testing forward error correction.

Also useful to have (and also implemented in the above repo) is a Markov source. This source is a Markov chain that goes from the idle state to the active state with probability p_start, and goes from the active state to the idle state with probability p_stop. In the idle state, it outputs zero. In the active state, it outputs 0 with probability 0.5 and 1 with probability 0.5

For an explanation of this noise model, see:
https://ntrs.nasa.gov/api/citations/19810024821/downloads/19810024821.pdf
R.L. Miller, L. J. Deutsch, S.A. Butman, On the Error Statistics of Viterbi Decoding and the Performance of Concatenated Codes. NASA Jet Propulsion Laboratory, 1981.

### Feature Urgency

medium (would be nice to have in the near future)

michael-roe, 2022-04-30:
P.S. I've already signed the FSF paperwork for contributing to gnuradio, so feel free to merge this code into the core if you want to.

luzpaz, 2023-10-11:
Any traction on this ?

willcode, 2023-10-11:
On what?

jasonuher, 2023-11-08:
For the majority of cases I think the 'random uniform source' (in gr-analog) should work.  Just set the minimum/maximum to 0 and 1.  Is there a good use case to make this a new block?