gnuradio/gnuradio #5798: gr-audio does not work with pulseaudio

catkira, 2022-05-01:
### What happened?

I compiled the latest gnuradio from this repo. I am using pulseaudio in WSL2 Ubuntu 20.04.
pulseaudio itself is working, I can play music with cmus and pavucontrol also works. 
However when I use the "Audio Sink" block in GNU Radio with setting Device Name = pulse
I get an error.

As a workaround I can use /dev/dsp for the audio sink and run
```
 padsp gnuradio-companion gr-bladeRF/apps/fm_receiver.grc
```
but thats not ideal and has a lot of latency in sound, because of the additional buffer being used.

### System Information

OS: Ubuntu 20.04 in WSL2
GR Installation Method: latest compiled from source (v3.11.0.0git-128-gfa8953de (Python 3.8.13))

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.11.0.0git-128-gfa8953de

### Steps to Reproduce the Problem

I am running the example that comes with gr-bladerf
gnuradio-companion gr-bladeRF/apps/fm_receiver.grc

### Relevant log output

```shell
audio_oss_sink :error: opening device pulse: No such file or directory
Traceback (most recent call last):
  File "/mnt/d/git/gr-bladeRF/apps/fm_receiver.py", line 322, in <module>
    main()
  File "/mnt/d/git/gr-bladeRF/apps/fm_receiver.py", line 300, in main
    tb = top_block_cls()
  File "/mnt/d/git/gr-bladeRF/apps/fm_receiver.py", line 205, in __init__
    self.audio_sink_0 = audio.sink(48000, 'pulse', True)
RuntimeError: audio_oss_sink

>>> Done (return code 1)
```

nils-werner, 2022-05-06:
I have had similar problems when using GNURadio in Conda environments with `pulseaudio` installed. It turned out I also needed to install `alsa-plugins`. Maybe something similar applies here?

catkira, 2022-07-01:
it works for me when I use padsp like this
```
padsp gnuradio-companion
```

luzpaz, 2023-07-13:
Is this still relevant ?

catkira, 2023-07-13:
can be closed

closed 2023-07-13 by catkira