gnuradio/gnuradio #5802: GRC crash after disabling several selected blocks

karolalexandersecret, 2022-05-02:
### What happened?

GRC crash after disabling several selected blocks at once. 

Console say: "Segmentation fault (core dumped)"

### System Information

Running under ubuntu 21.10
GR 3.10

GRC installed with:
>sudo add-apt-repository ppa:gnuradio/gnuradio-releases
>sudo apt-get update
>sudo apt install gnuradio

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

To reproduce the bug, select all the blocks from the flowgraph except variables and the message strobe block, and click D key. GRC should crash.

Here the flowgraph used:
[pdu_lambda_chirp_demo.grc.txt](https://github.com/gnuradio/gnuradio/files/8606236/pdu_lambda_chirp_demo.grc.txt)

karolalexandersecret, 2022-05-02:
After some tests. Here a simpler way to reproduce it:
Just disable the "PDU lambda" block in the following project:
[Bug.grc.txt](https://github.com/gnuradio/gnuradio/files/8606435/Bug.grc.txt)

mormj, 2022-05-03:
happens in ubuntu 20.04 as well.  haven't debugged, but able to reproduce easily as you describe.  Seems limited to the PDU lambda block

karolalexandersecret, 2022-05-03:
It also crash when only disabling the variable with N id.

Have tried to disable checkboxes in the view panel. Though it will not execute the python behind. But same result => crash

argilo, 2022-05-09:
The top of the C++ backtrace:
```
#0  0x00007ffff67e3697 in pmt::is_pair(std::shared_ptr<pmt::pmt_base> const&) (obj=std::shared_ptr<pmt::pmt_base> (empty) = {...})
    at /home/argilo/prefix_311/src/gnuradio/gnuradio-runtime/lib/pmt/pmt.cc:312
#1  0x00007ffff67f5f10 in pmt::write_list_tail(pmt::pmt_t, std::ostream&) (obj=std::shared_ptr<pmt::pmt_base> (empty) = {...}, port=...)
    at /home/argilo/prefix_311/src/gnuradio/gnuradio-runtime/lib/pmt/pmt_io.cc:31
#2  0x00007ffff67f51ba in pmt::write(std::shared_ptr<pmt::pmt_base>, std::ostream&) (obj=std::shared_ptr<pmt::pmt_base> (use count 5, weak count 0) = {...}, port=...)
    at /home/argilo/prefix_311/src/gnuradio/gnuradio-runtime/lib/pmt/pmt_io.cc:66
#3  0x00007ffff67f4451 in pmt::operator<<(std::ostream&, std::shared_ptr<pmt::pmt_base>)Python Exception <class 'gdb.error'>: value has been optimized out
 (os=..., obj=) at /home/argilo/prefix_311/src/gnuradio/gnuradio-runtime/lib/pmt/pmt_io.cc:109
#4  0x00007ffff67f64bb in pmt::write_string[abi:cxx11](std::shared_ptr<pmt::pmt_base>) (obj=std::shared_ptr<pmt::pmt_base> (use count 5, weak count 0) = {...})
    at /home/argilo/prefix_311/src/gnuradio/gnuradio-runtime/lib/pmt/pmt_io.cc:116
#5  0x00007fffb6330aa8 in operator() (__closure=<optimized out>, p=std::shared_ptr<pmt::pmt_base> (use count 5, weak count 0) = {...})
    at /home/argilo/prefix_311/src/gnuradio/gnuradio-runtime/python/pmt/bindings/pmt_python.cc:34
#6  pybind11::detail::argument_loader<std::shared_ptr<pmt::pmt_base> const&>::call_impl<std::__cxx11::basic_string<char>, bind_pmt(pybind11::module&)::<lambda(const pmt_t&)>&, 0, pybind11::detail::void_type> (f=<optimized out>, this=0x7fffffffaea0) at /usr/include/pybind11/cast.h:1207
#7  pybind11::detail::argument_loader<std::shared_ptr<pmt::pmt_base> const&>::call<std::__cxx11::basic_string<char>, pybind11::detail::void_type, bind_pmt(pybind11::module&)::<lambda(const pmt_t&)>&>
    (f=<optimized out>, this=0x7fffffffaea0) at /usr/include/pybind11/cast.h:1179
#8  operator() (__closure=0x0, call=...) at /usr/include/pybind11/pybind11.h:232
#9  _FUN(pybind11::detail::function_call&) () at /usr/include/pybind11/pybind11.h:210
#10 0x00007fffb6340660 in pybind11::cpp_function::dispatcher(_object*, _object*, _object*) (self=<optimized out>, args_in=(<pmt.pmt_python.pmt_base at remote 0x7fff8aa6f4f0>,), kwargs_in=0x0)
    at /usr/include/pybind11/pybind11.h:835
```
Python backtrace:
```
Traceback (most recent call first):
  <built-in method __str__ of PyCapsule object at remote 0x7ffff6aecae0>
  File "/home/argilo/prefix_311/lib/python3.10/site-packages/gnuradio/grc/gui/canvas/param.py", line 164, in format_block_surface_markup
    is_evaluated = self.value != str(self.get_evaluated())
  File "/home/argilo/prefix_311/lib/python3.10/site-packages/gnuradio/grc/gui/canvas/block.py", line 182, in <listcomp>
    markups = [param.format_block_surface_markup()
  File "/home/argilo/prefix_311/lib/python3.10/site-packages/gnuradio/grc/gui/canvas/block.py", line 182, in create_labels
    markups = [param.format_block_surface_markup()
  File "/home/argilo/prefix_311/lib/python3.10/site-packages/gnuradio/grc/gui/canvas/flowgraph.py", line 503, in create_labels
    element.create_labels(cr)
  File "/home/argilo/prefix_311/lib/python3.10/site-packages/gnuradio/grc/gui/canvas/flowgraph.py", line 195, in update
    self.create_labels()
  File "/home/argilo/prefix_311/lib/python3.10/site-packages/gnuradio/grc/gui/Application.py", line 110, in flow_graph_update
    fg.update()
  File "/home/argilo/prefix_311/lib/python3.10/site-packages/gnuradio/grc/gui/Application.py", line 235, in _handle_action
    flow_graph_update()
  File "/usr/lib/python3/dist-packages/gi/overrides/Gio.py", line 42, in run
    return Gio.Application.run(self, *args, **kwargs)
  File "/home/argilo/prefix_311/lib/python3.10/site-packages/gnuradio/grc/main.py", line 89, in main
    sys.exit(app.run())
  File "/home/argilo/prefix_311/bin/gnuradio-companion", line 96, in run_main
    sys.exit(main())
  File "/home/argilo/prefix_311/bin/gnuradio-companion", line 103, in <module>
    run_main()
```

argilo, 2023-11-19:
I did a bit more digging, and I think what's happening is this:

* The "PDU Lambda" block includes `import numpy as np`.
* Disabling it removes the import, after which the `vec` variable cannot be evaluated.
* The "Message PMT" parameter of the Message Strobe block becomes `pmt.cons(pmt.PMT_NIL, None)`, which results in a PMT pair with cdr set to a null pointer.
* Python attempts to convert this PMT to a string, which ultimately crashes when the null pointer is dereferenced.

I'm not sure where the correct fix belongs, but I would guess that either PMT construction should have some input validation, or the PMT library's string conversion should have better error handling.

argilo, 2023-11-19:
To reproduce the same crash outside GRC:

```
~$ python3
Python 3.10.12 (main, Jun 11 2023, 05:26:28) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pmt
>>> pmt.cons(pmt.PMT_NIL, None)
Segmentation fault (core dumped)
```

argilo, 2023-11-19:
This fix seems to work:
```diff
diff --git a/gnuradio-runtime/python/pmt/bindings/pmt_python.cc b/gnuradio-runtime/python/pmt/bindings/pmt_python.cc
index 33e345133b..488c8fc366 100644
--- a/gnuradio-runtime/python/pmt/bindings/pmt_python.cc
+++ b/gnuradio-runtime/python/pmt/bindings/pmt_python.cc
@@ -295,7 +295,7 @@ void bind_pmt(py::module& m)
     m.def("is_pair", &::pmt::is_pair, py::arg("obj"), D(is_pair));
 
 
-    m.def("cons", &::pmt::cons, py::arg("x"), py::arg("y"), D(cons));
+    m.def("cons", &::pmt::cons, py::arg("x").none(false), py::arg("y").none(false), D(cons));
 
 
     m.def("dcons", &::pmt::dcons, py::arg("x"), py::arg("y"), D(dcons));
```

If this is the right fix, then `.none(false)` should probably be added to all the arguments that are PMTs.

argilo, 2023-11-19:
It looks like Swig disallows `None`, so this isn't a problem in GNU Radio 3.8:
```
$ python3
Python 3.10.12 (main, Jun 11 2023, 05:26:28) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pmt
>>> pmt.cons(None, None)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/argilo/prefix_38/lib/python3/dist-packages/pmt/pmt_swig.py", line 2489, in cons
    return _pmt_swig.cons(x, y)
ValueError: invalid null reference in method 'cons', argument 1 of type 'pmt::pmt_t const &'
>>> pmt.car(None)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/argilo/prefix_38/lib/python3/dist-packages/pmt/pmt_swig.py", line 2497, in car
    return _pmt_swig.car(pair)
ValueError: invalid null reference in method 'car', argument 1 of type 'pmt::pmt_t const &'
```

marcusmueller, 2023-11-19:
> If this is the right fix, then .none(false) should probably be added to all the arguments of type pmt_t.

looks like it. 

Assuming I'm right about PMT-held data pointers being immutable:

We the `const char*`-wrapping PMT `mp` (and underlying `string_to_symbol`) should flat out reject a null pointer. That is never a correct symbol, and due to immutability, that could never change, so that is guaranteed to produce a broken PMT.

closed 2023-11-20 by willcode