gnuradio/gnuradio #5818: qa_crc16_async_bb is flaky

argilo, 2022-05-06:
The `qa_crc16_async_bb` test added in #5700 has already failed three times in CI:

* https://github.com/gnuradio/gnuradio/runs/6312845070?check_suite_focus=true
* https://github.com/gnuradio/gnuradio/runs/6315190144?check_suite_focus=true
* https://github.com/gnuradio/gnuradio/runs/6316453166?check_suite_focus=true

```
 FAIL: test_crc16 (__main__.qa_crc16_async_bb)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "$SRC_DIR/gr-digital/python/digital/qa_crc16_async_bb.py", line 48, in test_crc16
    self.assertEqual(dbg_check.num_messages(), 1)
AssertionError: 0 != 1
```

The failure looks similar to the one corrected in #5754, where messages don't make it through before the flow graph shuts down.

So far I haven't been able to reproduce a failure locally. If the test continues to be problematic, it may be best to remove it until the source of the problem is tracked down.

michael-roe, 2022-05-06:
This seems to be a fundamental correctness problem with tests of async message blocks.

It looks like we could do with a mechanism for terminating the test when the message debug block has received a message. (Otherwise there's a race condition between the test ending and the message being received).

argilo, 2022-05-06:
I expected #5754 to fix that, but there could be more to the problem.

argilo, 2022-05-06:
Mystery solved. The Async CRC16 block was broken: #5830.

PR #5830 (2022-05-11): digital: Fix CRC checking in Async CRC16 and Async CRC32 blocks
## Description
The Async CRC16 and Async CRC32 blocks can perform misaligned loads, as was the case in #5807.

In addition, the Async CRC16 check fails randomly because it reads four bytes from memory instead of two, causing #5818.

I've fixed both problems by using `memcmp` (with the correct number of bytes) to perform the CRC comparison.

I also removed the unnecessary `crc16_async_bb_impl::process_crc` method to make the implementations of Async CRC16 and CRC32 nearly identical.

## Related Issue
* Fixes #5818.

## Which blocks [... cut]
closed 2022-05-11 by mormj