gnuradio/gnuradio #5841: MacOS: GNU Radio only 1/4 the size of the GTK window

fah, 2022-05-08:
### What happened?

GTK windows appears with only top left quarter is filled with gnu-radio screen.
[image]

----

I tried the following patch:
```
sudo vim /usr/local/lib/python3.9/site-packages/gnuradio/grc/gui/DrawingArea.py

  def _update_size(self):
        w, h = self._flow_graph.get_extents()[2:]
        """
        PATCH maually applied: https://github.com/conda-forge/gnuradio-feedstock/pull/85/files
        OLD:
        self.set_size_request(w * self.zoom_factor + 100, h * self.zoom_factor + 100)
        """
        scale_factor = self.get_scale_factor()
        self.set_size_request(
            w * scale_factor * self.zoom_factor + 100,
            h * scale_factor * self.zoom_factor + 100,
        )

```
Which didn't change anything.

### System Information

Installing GNU Radio via  
[CondaInstall - GNU Radio](https://wiki.gnuradio.org/index.php/CondaInstall)
on an iMac with native Graphic set to normal size. Same happens with other resolutions, too.

Versions:
MacOS 11.6.4
GNU Radio Companion 3.10.2.0 
conda 4.12.0

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.2.0

### Steps to Reproduce the Problem

1. Start ```gnuradio-companion``` from commandline
2. GTK windows appears with **only top left quarter is filled with gnu-radio screen**.
3. Click on a menu -> crash with log output.  

### Relevant log output

```shell
Log:

(gnuradio) x@iMac Project % gnuradio-companion
>>> Warning: vocoder_codec2_decode_ps - option_attributes are for enums only, ignoring
>>> Warning: vocoder_codec2_encode_sp - option_attributes are for enums only, ignoring
<<< Welcome to GNU Radio Companion 3.10.2.0 >>>

Block paths:
	/Users/x/.grc_gnuradio
	/Users/x/opt/anaconda3/envs/gnuradio/share/gnuradio/grc/blocks

Loading: "/Users/x/radio_3.9.grc"
>>> Done

Loading: "/Users/x/Downloads/Gnuradio/GNU Radio_samples/wbfm with audio out.grc"
>>> Done

(gnuradio-companion:42427): Gtk-WARNING **: 18:59:49.996: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node header owner gnuradio+grc+gui+Notebook+Notebook)

(gnuradio-companion:42427): Gtk-WARNING **: 18:59:49.996: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node tabs owner gnuradio+grc+gui+Notebook+Notebook)

(gnuradio-companion:42427): Gtk-WARNING **: 18:59:49.996: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node tab owner gnuradio+grc+gui+Notebook+Notebook)

(gnuradio-companion:42427): Gtk-WARNING **: 18:59:49.996: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node tab owner gnuradio+grc+gui+Notebook+Notebook)
Assertion failed: (!surface->finished), function _cairo_surface_release_source_image, file cairo-surface.c, line 2092.
zsh: abort      gnuradio-companion
```
```

ryanvolz, 2022-05-09:
Since the error message includes something about a cairo surface, and I know the conda package contains a patch that has to do with Cairo's ImageSurface and may be causing problems (even though at one time it was supposed to fix a problem), would you mind trying an alternative version without the patch? You can install from my personal channel with the `no-imagesurface-patch` label:
```
conda install -c ryanvolz/label/no-imagesurface-patch gnuradio
```
(or I'd suggest you use `mamba` instead of `conda` if you can). Maybe it does nothing or re-activates an old bug, but it would be good to know.

Other than that, I think your error message at least has a few hints as to where to look for what is causing the problem. Hopefully someone who knows more than me about GRC+gtk can take a look. It would be helpful to confirm the following, though: does your iMac have a high-DPI (Retina) display? I think the answer is "no" based on your description, but maybe I'm misunderstanding.

ryanvolz, 2022-05-09:
Also, to reverse that scaling patch, the correct place to do it would be `/Users/x/opt/anaconda3/envs/gnuradio/lib/python3.9/site-packages/gnuradio/grc/gui/DrawingArea.py` so it actually affects the conda installation, and not in `/usr/local` like your message says. Maybe that's what you actually did, but it's good to check.

fah, 2022-05-09:
@ryanvolz 
Resolution:
I have an iMac (Retina 5K, 27-inch, 2017).
What I ment with the resolution is that I can change the **magnification** in 5 steps where the middle is called "normal". 
I tried the other magnifications too and it had no influence. Strangely, when I moved the GTK window to a second screen with a lower resolution the GTK window was filled but the app crashed when I touched the left menues, too.

Old patch:
The folder is /Users/x/opt/anaconda3/envs/gnuradio/lib/python3.**10**/site-packages/gnuradio/grc/gui/DrawingArea.py 
Initialy I indeed did reverse the patch at the right place. Nothing changed. I added an error to see that this was the right place. :-)

Your imagesurface patch:
I applied your patch
... it's currently running I'll update later ...

fah, 2022-05-10:
@ryanvolz 
... 5 hours later the patch was done. With conda. I guess next time I use mamba. :-)
Nothing changed.

Should I link this bug in Cairo ImageSurface?

fah, 2022-05-10:
@ryanvolz 
Something changed: On the 5k main screen it crashes. But when I open the screen on the main screen and send the gkt window (with right click on the enlarge button of the gtk window) to the second screen with a lower resolution, I can work now!

Without your patch the app crashed there, too.

ryanvolz, 2022-05-10:
Interesting. The original patch was meant to fix #2726, so I wonder if you have problems when connecting two blocks now? Clearly it's better to not crash though, so I guess I will drop that patch officially and see how it goes.

As for the issues with the high-DPI display, it sounds related to the issues reported in #5613. Maybe there are some helpful workarounds in there, but clearly the scaling is still causing issues.

fah, 2022-05-10:
@ryanvolz 
I would not apply that patch made for #2726:

I can connect two and more blocks. 

After starting a graph, the graph is actually running, I can close the graph, BUT I can't work in gnuradio anymore afterwards, because it is unresponsive. Shows turning ball. No reaction to the mouse. I have to kill it.

I only can give you a screenshot:
[image]

What I noticed, too:
I use HackRF One & PortaPack (GreatScott) and the graph (osmocom Source -> (WBFM Receive -> Audio Sink), QT GUI Sink  ) seems not to clean up after ending. RX Led on HackRF is off but the screen of PortaPack still shows the setup. 

It is a little frustrating to restart GNU Radio on each iteration.
---

I don't see any relation to #5613. I have no M1 and I did'nt need to change the magnification of the 5k screen to get it running.

fah, 2022-05-11:
Possibly 
Duplicate of #5085

fah, 2022-05-11:
With ```gnuradio-companion --log debug``` I get, when I click the menues on the 5k display:
```log
zsh: bus error  gnuradio-companion --log debug
```

fah, 2022-05-11:
Possible relation to https://gitlab.freedesktop.org/cairo/cairo/-/issues/420

fah, 2022-05-11:
https://gitlab.freedesktop.org/cairo/cairo/-/issues/568 created

ryanvolz, 2022-05-11:
> Possible relation to https://gitlab.freedesktop.org/cairo/cairo/-/issues/420

This looks like it would be the reason that the patch to fix #2726 causes crashes on Big Sur (since it introduces a buggy cairo surface that otherwise wouldn't be involved). Maybe it is also responsible for other crashes. Unfortunately it seems like we're running into multiple bugs here:

1) Crash related to how cairo surfaces are handled on macOS Big Sur with the stable cairo release 0.16.0
2) Crash when interacting with the upper left menus on macOS (Big Sur specific? same cause as 1?)
3) GRC window is 1/4 size on high DPI displays (maybe sometimes fixed by #5111, but clearly still happens under some circumstances)

Anything I'm missing? I'll try to get the cairo patch that fixes https://gitlab.freedesktop.org/cairo/cairo/-/issues/420 merged into conda-forge; I see that Homebrew includes it.

ryanvolz, 2022-05-11:
The patched cairo that fixes https://gitlab.freedesktop.org/cairo/cairo/-/issues/420 is now installable from my channel:
```
conda install -c ryanvolz/label/bigsur-patch cairo
```
@fah See if that helps with the crashes. I'll get it merged into conda-forge if you give a positive report.

fah, 2022-05-12:
>     1. Crash related to how cairo surfaces are handled on macOS Big Sur with the stable cairo release 0.16.0
If that also means that hovering over a the tabs close button causes a cash? 
I think if you keep several tabs (>2) open, this error is amplified.

---

Last night I reinstalled the devtools of XCode and ran a "brew upgrade". This improved the stability a little.

So I was able to work with GR but had some crashes with the blocks, file dialog (opening a directory that was accessible) and dropdown elements in blocks dialog itself, too.
The crashes where not always reproducable. After restart most times it worked well.

---

I would suggest offering a version that includes the debug information. So that a crash can be investigated. I don't know how to do that for the wrong parts only or for all parts.

---

I am currently applying your patch "ryanvolz/label/bigsur-patch cairo" ...

ryanvolz, 2022-09-19:
Recently I was able to observe a crash in CI testing of the conda package caused by while opening the File menu in GRC, and I can confirm that the `cairo` patch affecting macOS Big Sur+ mentioned above fixes it. That patch [has now been applied](https://github.com/conda-forge/cairo-feedstock/pull/68) to the conda-forge version of `cairo`. Fingers crossed that it solves more of these crashes.

ryanvolz, 2022-11-10:
Any chance https://github.com/gnuradio/gnuradio/pull/6352 fixes this? That's a python-only fix, so you can manually edit the appropriate installed python files to test it.

marcusmueller, 2024-01-31:
haven't heard from this fundamental UI error in more than a year after asking for whether it was solved, so let's assume it was solved.

closed 2024-01-31 by marcusmueller