gnuradio/gnuradio #5851: Decode CCSDS 27 degrades over time

michael-roe, 2022-05-15:
### What happened?

I encoded a stream of bits with Encode CCSDS 27, conveted 0 and 1 to (float) -1 and +1, added Gaussian noise (amplitide 0.69) and decoded with Decode CCSDS 27.

It works for about the first 10 million bits, and then the bit error rate of the decoded bits gets dramatically worse.

(I have a vague memory that we had a bug like this before).

What I expect to happen is that the bit error rate in the output should remain approximately the same, no matter how many bits have previously been processed.

As a work-around, I can probably use one of the other implementations of the Viterbi algorithm in Gnuradio (there are several). If this is a known limitation, it probably ought to be documented.

### System Information

OS: Ubuntu 22.04
GR Installation Method: compiled from source

### GNU Radio Version

3.11-git (main)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

I encoded a stream of bits with Encode CCSDS 27, conveted 0 and 1 to (float) -1 and +1, added Gaussian noise (amplitide 0.69) and decoded with Decode CCSDS 27.

michael-roe, 2022-05-15:
See also:
https://destevez.net/2017/07/degradation-bug-in-gnu-radio-decode-ccsds-27/

michael-roe, 2022-05-15:
The context in which I originally encountered the bug involved 56 kbps digital voice (common in telephony) being CCSDS encoded for transmission. (Yes, I know, there are voice codecs with better compression). At that data rate, you encounter the bug if the voice call lasts longer than a couple of minutes, which is quite likely to happen.

Which is to say: failing after 1E7 bits is likely to impact real applications.

Mixaill, 2022-05-18:
@michael-roe 

Thank you a lot, struggling with the decoding issues for a week.

We encountered this bug when decoding CCSDS TM packets (~10% of corrupted bytes after several 10^7 symbols or so). 

Replacing `Decode CCSDS 27` with the combination of `FEC Extended Decoder` and `CC Decoder Definitions` fixes the problem.

argilo, 2022-10-09:
I tracked down the bug. The Viterbi path metrics are never reset, so they eventually overflow. This should take a long time, since the metrics are stored in `long` variables:

https://github.com/gnuradio/gnuradio/blob/c49548c2e079fa3b50efbd8d1699b5089d523438/gr-fec/include/gnuradio/fec/viterbi.h#L23-L26

However, the `BUTTERFLY` macro uses `int` for its calculations, truncating the metrics:

https://github.com/gnuradio/gnuradio/blob/c49548c2e079fa3b50efbd8d1699b5089d523438/gr-fec/lib/viterbi/viterbi.cc#L50-L74

Changing `m0` and `m1` to `long` solves the problem.

PR #6234 (2022-10-20): fec: Prevent integer overflow in path metric calculations
## Description
The Decode CCSDS 27 block fails after processing approximately 4.2 million input bits. This happens because Viterbi path metric calculations are done using signed 32-bit integers, even though they are stored in signed 64-bit integers. Switching the calculations to signed 64-bit integers resolves the issue.

## Related Issue
Fixes #5851.

## Which blocks/areas does this affect?
* Viterbi decoder

## Testing Done
I ran @daniestevez's test script (https://destevez.net/2017/07/degradation-bug-in-gnu-radio-decode-ccsds-27/) an [... cut]
closed 2022-10-20 by mormj