gnuradio/gnuradio #5891: Remove Boost dependency

argilo, 2022-05-25:
One of the goals of GNU Radio 4.0 is to remove Boost as a dependency. After the open Boost removal PRs (#5706, #5846, #5852, #5853, #5888, #5890) are merged, there will only be a small amount of Boost left. The table below summarizes the remaining work.

Boost component | Used in | LOC | Proposed replacement | Notes
------------------------|------------|-------:|--------------------------------|---------
Boost.Test | C++ test cases | 921 | ???
Boost.Asio | gr-network, rpcserver_booter_thrift | 91 | [Stand-alone ASIO library](https://think-async.com/Asio/)
Boost.Any | PMT | 27 | `std::any` | See #5218, #5311
Boost.Thread | Thread-per-block scheduler | 26 | `std::thread` | Needs refactoring to remove `interrupt()`
Boost.Program_options | `gr::prefs`, gnuradio-config-info, UHD tags_demo, GRC C++ generation | 4 | ???
`file_lock` | gr-fft | 2 | ???
`this_line_location` | `gr::prefix()` | 1 | Copy code from Boost? ([Source](https://github.com/boostorg/dll/blob/develop/include/boost/dll/runtime_symbol_info.hpp))
`math::constants::pi` | GRC C++ generation | 1 | `3.14159265358979323846` or `std::numbers::pi` (C++20)

argilo, 2022-05-25:
Thanks for the suggestions! I've added them to the table.

> Boost.program_options --> cli11 works well and is a minimal dependency

This will take care of the CLI parts, but `gr::prefs` also uses program_options to parse config files (which use INI format). We'll need a replacement for that as well.

marcusmueller, 2022-05-25:
I think the most intrusive change is going to be the boost::thread replacement. Had a go at that on two occasions, and I think it leaked into the API. So, yeah, touching the core tpb thread body would be necessary, which might not be much fun (as it's possible to introduce new shutdown bugs), but maybe it's easy, in the end an interrupt point is just a semaphore "should this thread still be running? If not, throw an exception!", and maybe just replacing that with a check against one of the already existing "done" state indicators would actually untangle the code, and force `sched_tpb::stop()` to actually signal shutdown to all threads instead of generally interrupting them, which might or might not wake them, and make them finish their job.

mbr0wn, 2022-05-25:
I see no value in replacing Boost.Test and Boost.Program_options.

argilo, 2022-05-25:
> in the end an interrupt point is just a semaphore "should this thread still be running? If not, throw an exception!"

The annoying part is that there are predefined interruption points in a bunch of places, notably `boost::condition_variable::wait()`. The full list is here:

https://www.boost.org/doc/libs/1_79_0/doc/html/thread/thread_management.html#thread.thread_management.tutorial.interruption.predefined_interruption_points

When I tried to remove Boost from the thread-per-block scheduler, deadlocks occurred during shutdown while waiting for the `output_cond` condition variable. We'd need to make sure condition variables are signaled during shutdown.