gnuradio/gnuradio #5919: Installation of OOT module on MacOS fails because of corrupted path in cmake

dxxx9, 2022-06-03:
### What happened?

A bug happened!

Installation of own OOT module fails because of a wrong path. See the log output below. Somehow it's a relative path, but mixed with the `CMAKE_INSTALL_PREFIX` in front of it.

That's the conflicting `line 41` of `python/cmake_install.cmake`:
```
file(INSTALL DESTINATION "${CMAKE_INSTALL_PREFIX}/../../../../../../Cellar/gnuradio/3.10.2.0/libexec/venv/lib/python3.9/site-packages/<my-module>" TYPE FILE FILES
```

### Workaround | Quickfix
Fixed this issue by navigating to `python/cmake_install.cmake` file and correcting the file path in two places. Then the installation worked fine.

### System Information

> OS: Mac
GR Installation Method: brew
cmake version 3.23.2

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.2.0

### Steps to Reproduce the Problem

Follow [the guide](https://wiki.gnuradio.org/index.php/OutOfTreeModules) to install a custom OOT module.

```
cmake ../               ✅ 
make                    ✅ 
make test               ✅ 
sudo make install       ❌ 
```
 

### Relevant log output

```shell
CMake Error at python/cmake_install.cmake:41 (file):
  file cannot create directory:
  /usr/local/../../../../../../Cellar/gnuradio/3.10.2.0/libexec/venv/lib/python3.9/site-packages/<my-module>.
  Maybe need administrative privileges.
Call Stack (most recent call first):
  cmake_install.cmake:51 (include)
```
```

JayKickliter, 2022-06-18:
I can confirm this bug.

ryanvolz, 2022-11-10:
Perhaps this was fixed by #5979?

marcusmueller, 2024-01-31:
no reaction in > 1 year after asking back :( also, might very realistically have been a brew packaging problem into which we have limited insight :(

closed 2024-01-31 by marcusmueller