gnuradio/gnuradio #5923: Provide Flatpak

noonien-d, 2022-06-03:
### Feature Description

The GNURadio package in repositories tends to be outdated and compiling by source is a struggle due to many dependencies. 

Flatpak bundles all dependencies and its also possible to install a recent version on an older system. Nowadays, Flatpaks are supported by most Linux distributions and you need only one Flatpak for all distributions.

As a starter, I managed to create a working Flatpak [manifest for maint-3.10](https://github.com/noonien-d/flatpak-gnuradio ), but it doesn't contain all modules yet.

When the manifest reaches a good state, one might publish it on [flathub](https://flathub.org/), which is the best known repository for Flatpaks. 

- [About Flatpak](https://flatpak.org/)
- [Flatpak FAQ](https://flatpak.org/faq/)
- [App submission on Flathub](https://github.com/flathub/flathub/wiki/App-Submission)

### Feature Urgency

low (just an idea)

marcusmueller, 2022-06-07:
> The GNURadio package in repositories tends to be outdated and compiling by source is a struggle due to many dependencies.

right now, we're pretty good! The current Fedora, Ubuntu, debian, mint, manjaro... all ship the latest minor release!

> Flatpak bundles all dependencies and its also possible to install a recent version on an older system. Nowadays, Flatpaks are supported by most Linux distributions and you need only one Flatpak for all distributions.

Flatpaks are also not a proper way of distributing a library for other software to use, as far as I can tell. So, we're aiming at the "GNU Radio Companion itself without Out-of-Tree modules" use case here – which is an important one, especially in educational settings.

I'd, however, this would be a "gnuradio-companion" flatpak – GNU Radio is simply more than that! (One could argue GNU Radio is just a library, and the signal processing applications you build from it could reasonably be considered "apps", take gqrx as an example.)

> As a starter, I managed to create a working Flatpak [manifest for maint-3.10](https://github.com/noonien-d/flatpak-gnuradio), but it doesn't contain all modules yet.

That sounds like a lot of work, thank you! Would you be able to maintain that, going forward? That would be awesome. Which modules are currently built? 

Another question, as you seem to be knowledgeable about flatpaks: How well does the integration with libusb bulk transfer USB devices work? How does one integrate hardware drivers in the flatpak ecosystem? I know there's some way of defining dependency trees, but I frankly don't understand it.

I'm asking because to make GNU Radio really useful for a wide audience (and not add to the confusion by adding a distribution method that carries the name, but doesn't let beginner users do what they came to do), it'd be desirable to integrate at least with UHD (the USRP hardware driver) and RTL-SDR by means of  librtlsdr, and ship libsoapy as middleware.

noonien-d, 2022-06-08:
There are cases where it is not possible to install a recent Linux distribution, like in companies. For example I need to use a RedHat based distribution because of some commercial software support. 

For some people it might also be helpful that you can install flatpaks per-user, without superuser permissions.

You are right, its more like a companion flatpak, you are not able to use the framework in other applications. A possible gqrx flatpak needs to ship its own copy of GNURadio. 

USB devices shouldn't be a problem, access depends on the sandbox settings. As long as drivers are just user space libraries, they could be added to the flatpak container. 

Out of tree modules could be shipped as flatpak extension (but they need to get bundled accordingly).

I'll post a list of enabled modules after the next rebuild.

Po84210, 2022-06-11:
I’d like to have a Flatpak version of GNU Radio, to have more control over dependencies on the system and also device permissions.

marcusmueller, 2022-06-11:
@Po84210  thanks for weighing in on this! I don't think flatpak addresses device permissions, though: The user executing the flatpak needs to have the same permissions as if the UHD/rtl-sdr/… driver was run by them "natively".

MBasala, 2022-10-03:
I have created a Flatpak for personal use. If any Devs want to improve it and submit to flat hub I would be happy with that.

My GNURadio Flatpak: https://github.com/MBasala/GNURadio-Flatpak

BloodBlight, 2023-01-11:
My two cents on this...

I am rebuilding one of my laptops right now, just for a GNU Radio install I want to try.  It has been a few years, but last time I tried to install it...  It was a nightmare!  And some of the apps I wanted to use wanted different, incompatible versions of GNU Radio (I think one would ONLY build on Ubuntu 16 or 18, can't remember).  Having a good template that could be easily added to for these type of projects would be AMAZING!  Especially for older projects that are no longer being maintained.

Sorry this comment is so light on details, I am an extreme novice when it comes to GNU Radio and it really was years ago at this point, and may not be relevant anymore.

agentgerbil, 2023-06-16:
I'd like to see a flatpak so I can install it on my Steam Deck.

MBasala, 2023-06-16:
> I'd like to see a flatpak so I can install it on my Steam Deck.

I used mine on my steam deck and got it working. Mind you depending on your SDR you may need to still install some low level drivers since flatpak doesn't do that.

> I have created a Flatpak for personal use. If any Devs want to improve it and submit to flat hub I would be happy with that. It has almost every module working and a lot of SDR devices built in. HackRF, ADALM-PLUTO, RTL-SDR.. etc to name a few. It does include gr-osmosdr and soapySDR as well.
> 
> My GNURadio Flatpak: https://github.com/MBasala/GNURadio-Flatpak

marcusmueller, 2025-07-26:
@AsciiWolf No, fedora is not moving to flatpak for libraries, that's not what these are for; we're not responsible for third party packaging that has **explicitly** been declared for personal use only.

closed 2025-07-26 by marcusmueller

AsciiWolf, 2025-07-26:
I was talking primarily about GNU Radio Companion. And yes, Fedora is going to move away from RPMs to Flatpaks for user apps. It won't happen soon, probably not earlier than one or two years from now, but it already is planned.

marcusmueller, 2025-07-26:
@AsciiWolf you don't seem to axtually read my replies. GNU Radio is not just an App, full stop, and as active member of Fedora's developer community I think I might have noticed if we abandoned RPM. Please lay it off for now, flatpak packaging a library that would require a whole slew of new xdg portals to be useful with the various things you would want GNU Radio to talk to is not on our priorities list.