gnuradio/gnuradio #5932: Help regarding dilemma

otpisani, 2022-06-06:
Sorry if I violate policy rules, but can I ask question regarding idea that I like to solve, or point me on some forum regarding gnuradio comapnion. 

Can I use companion to scan some bandwith region for some modulation frequency. Gnuradio need to compare wav or raw files within database or files on disk, with frequency modulation which he will discover when scanning some frequency bandwith. Companion need to compare 2-3 frequency modulation, using basic statement like IF, AND, OR. Basically I need idea how to solve the issue.

willcode, 2022-06-06:
Please ask for help on chat.gnuradio.org or the mailing list. The issue tracker is used for bug reports and feature requests.

closed 2022-06-06 by willcode

otpisani, 2022-06-06:
Thanks man.

Lp, Robin

*Mentally strong people DO NEVER react.*
They don't allow others to control their behavior by pushing them to react to incidents and rough words.  They step back, observe, analyse the situation and think before they utter a word.  They are aware of what they do, what they say, and how they feel.