gnuradio/gnuradio #5948: qa_constellation_receiver is flaky

argilo, 2022-06-08:
The qa_constellation_receiver test recently failed on a maint-3.9 backport (#5945):

https://github.com/gnuradio/gnuradio/runs/6772627220?check_suite_focus=true

The failure occurred on Fedora 36, and the details were as follows:
```
======================================================================
FAIL: test_basic (__main__.test_constellation_receiver)
Tests a bunch of different constellations by using generic
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/__w/gnuradio/gnuradio/gr-digital/python/digital/qa_constellation_receiver.py", line 133, in test_basic
    self.assertTrue(correct > req_correct)
AssertionError: False is not true
----------------------------------------------------------------------
Ran 2 tests in 0.877s
FAILED (failures=1)
Constellation is <gnuradio.digital.digital_python.constellation_expl_rect object at 0x7f37440399b0>. Differential is True.  Required correct is 0.8. Correct is 0.6714975845410628. FAIL.
```

So far I have not been able to reproduce the failure locally.

marcusmueller, 2022-06-08:
It seems to be deterministically seeded:
https://github.com/gnuradio/gnuradio/blob/main/gr-digital/python/digital/qa_constellation_receiver.py#L87-L92

argilo, 2024-01-04:
I suspect that the flaky test was due to the Polyphase Clock Sync bug fixed in #7040, so I'll close this off. If another test failure occurs, I'll re-open this.

closed 2024-01-04 by argilo