gnuradio/gnuradio #5959: analog/fastnoise_source constructor argument `seed` type inconsistent with docstring

bgottula, 2022-06-17:
### What happened?

The [docstring](https://github.com/gnuradio/gnuradio/blob/main/gr-analog/include/gnuradio/analog/fastnoise_source.h#L46-L48) for analog/fastnoise_source `seed` argument states:

> seed for random generators. Note that for uniform and Gaussian distributions, this should be a negative number.

This always seemed a strange requirement, but the type was changed in a recent commit from `long` to `uint64_t` so clearly a negative value is not even possible. I'm guessing the docstring is probably wrong, but I don't know the history of why a negative number was recommended previously.

Commit where the argument type was changed: https://github.com/gnuradio/gnuradio/commit/4dd02dac2fcaf45f398c074735a7359330f78d77

Related issue: https://github.com/gnuradio/gnuradio/issues/1559

I also note that the type of the `seed` parameter for the analog/noise_source block is still a `long` and it has the same docstring: https://github.com/gnuradio/gnuradio/blob/main/gr-analog/include/gnuradio/analog/noise_source.h#L44-L48

### System Information

OS: Ubuntu 20.04
GR Installation Method: apt via gnuradio-releases PPA

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.2.0

### Steps to Reproduce the Problem

Use negative seed value as documentation suggests when instantiating fastnoise_source. (This produced no error in GNU Radio 3.9)

### Relevant log output

```shell
In [12]: a = analog.fastnoise_source_c(analog.GR_GAUSSIAN, 1, -1, 100000)
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Input In [12], in <cell line: 1>()
----> 1 a = analog.fastnoise_source_c(analog.GR_GAUSSIAN, 1, -1, 100000)

TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. gnuradio.analog.analog_python.fastnoise_source_c(type: gr::analog::noise_type_t, ampl: float, seed: int = 0, samples: int = 16384)

Invoked with: noise_type_t.GR_GAUSSIAN, 1, -1, 100000
```

marcusmueller, 2022-06-17:
Chances are high it's me that changed the argument type. Yes, the docstring is incorrect!

And I still think we should change the fastnoise source's long to an uint64_t, but we can't do that on any of the released versions, only on the `main` branch, as it does break API and we can't do that for released versions!

So the desirable solution is twofold:
1. make a fix for the docstring against `main`. Say that this should be backported!
2. change long->uint64_t for the fastnoise source in a separate PR against `main`.

marcusmueller, 2024-01-31:
solved per #5988 and #5989