gnuradio/gnuradio #5968: RF64 support for the WAV File Source Block

pclov3r, 2022-06-26:
### Feature Description

Currently, The WAV File Source Block only supports WAV files up to 4 GB. As result of these limitations the RF64 standard was developed that is based on Microsoft RIFF/WAVE format that supports WAV files up to 16 EiB.

A few programs already support the RF64 format and output I/Q files wrapped around it. I figure in time more will support this format as 4 GB file limit for I/Q files is rather low especially at high sample rates. 

Currently, According to https://wiki.oarc.uk/gnuradiotips you can work around this by 

> For RF64 WAV files you can use a File Source followed by an iShort to Complex to effectively skip the header and send out valid data.

### Feature Urgency

medium (would be nice to have in the near future)

marcusmueller, 2022-06-26:
> If you attempt to use a RF64 file now you will get an error

That's not true, I just tried! We've been¹ using the libsndfile library in the wavfile source since GNU Radio 3.9.0.0; you, and the wiki page you link to, might be on an older version of GNU Radio:

[image]

```shell
~/flowgraphs/wav_to_audio.py -f ./untitled.rf64
# I hear the beep from the file
```
I'm attaching the test file for you to try yourself:

[wav_to_audio.zip](https://github.com/gnuradio/gnuradio/files/8986292/wav_to_audio.zip)

Now, does that say it works for files > 4GB? Can't say I've tried. Gimme a second, need to hand-craft a file like that.

----
¹ @drmpeg implemented that for GNU Radio 3.9.0.0

marcusmueller, 2022-06-26:
btw, the fixed-length cutting off of a header is probably incorrect, I think you can have variable-length metadata somewhere in the wav file, and also in rf64.

pclov3r, 2022-06-26:
@marcusmueller Thanks for replying!

I am using a older version of GNUradio 3.8.2 as that is what is provided by the distro I'm using.

Given it appears this is already implemented in GNUradio 3.9 and newer I'll go ahead and close this off.

closed 2022-06-26 by pclov3r

marcusmueller, 2022-06-26:
Hi @pclov3r! Still thanks for raising this, as such input is important to us knowing what people actually use. (In this case we're lucky, the fact that the new Wavfile sink even lets you select RF64 as a file type option is possibly because @drmpeg himself is a broadcasting technology expert.)

As said, haven't tested whether it works for actually reading > 4GB yet, just that RF64 is generally supported. I'm however [including the flow graph](https://github.com/gnuradio/gnuradio/files/8986337/generate_large_rf64_code.zip)) I've whipped together to generate a 4.9 GB RF64 file (because I had no better way) here, so that you have your own testing platform, together with the file reader from previous comments!

I'll also try (in the next comment) including a 4.9 GB (4.7 GiB) test-RF64 generated with that flow graph to allow you to test your own software workflows against it. Obviously, I can't attach 4.9 GB to an issue on github, so it's been ZSTD-compressed. You need zstd to uncompress the large.rf64.zstd (in the large.zip file) back to the original large.rf64. 

[image]

marcusmueller, 2022-06-26:
Here's a small 4.9 GB fille: 

[large.zip](https://github.com/gnuradio/gnuradio/files/8986362/large.zip)

pclov3r, 2022-06-26:
@marcusmueller Thanks again for the help! 

Hopefully down the road this issue will be easier for others to try find. I tired to look if RF64 was supported before raising this issue but couldn't exactly find if this was supported :)

marcusmueller, 2022-06-26:
you're right, we did not write this down appropriately. I made a small emergency edit to https://wiki.gnuradio.org/index.php?title=Wav_File_Source ; can you check whether it's appropriate? If you want to add some wordage, change anything, or so, please do feel free to just edit that wiki page (if you do it anonymously, it ends up in our review queue, if you register, you might need to wait for your account to be manually verified, but then you're out of review queue latency).

pclov3r, 2022-06-26:
@marcusmueller Slightly off topic but just noticed 
> other file format that [libsndfile](http://libsndfile.github.io/libsndfile/formats.html) supports

Does that mean the WAV file sink supports way more then just WAV files? 

If it can do FLAC files that would be a major benefit to compress I/Q files to save space.

marcusmueller, 2022-06-26:
yes, it does. See the updated wiki page for the wav file sink :)

pclov3r, 2022-06-26:
Hum, To me it seems like it should no longer be called WAV File Source/Sink anymore? Perhaps it should have a different name now. 

Just an idea unless I'm off course here :)

marcusmueller, 2022-06-26:
you're very right; I think there was also a discussion on this when #3600 was merged, just can't find it right now. Maybe open up a new issue that says "wavfile_sink, source: naming is confusing in GRC"?

The reason the code structure (the class) has not been renamed is drop-in compatibility; that doesn't mean we can't, at least in upcoming version (shouldn't break existing code) have something that is "visibly" called "Audio File Sink" instead.

pclov3r, 2022-06-26:
@marcusmueller I've gone ahead an did so https://github.com/gnuradio/gnuradio/issues/5970 :)