gnuradio/gnuradio #5971: Wavfile Sink: Header docstring still only mentions PCM, but we do all of libsnd

marcusmueller, 2022-06-26:
### What happened?

[A bug happened!](https://github.com/gnuradio/gnuradio/blob/v3.10.3.0/gr-blocks/include/gnuradio/blocks/wavfile_sink.h) still says 
https://github.com/gnuradio/gnuradio/blob/v3.10.3.0/gr-blocks/include/gnuradio/blocks/wavfile_sink.h#L22

Also, [`Check gr_make_wavfile_source() for extra info.`](https://github.com/gnuradio/gnuradio/blob/main/gr-blocks/include/gnuradio/blocks/wavfile_source.h#L26) is utterly meaningless (and would not be much better if this was still GNU Radio 3.6, where such a function might have still existed.

same for `…_source.h`

Now, starting since 3.9.0.0, we do all the formats that libsndfile can handle.

# Fix

* Remove https://github.com/gnuradio/gnuradio/blob/main/gr-blocks/include/gnuradio/blocks/wavfile_source.h#L26 
* replace the `\brief …` sentence with "Write samples to an Audio file"
* Add the list of supported audio containers from [the wiki page](https://wiki.gnuradio.org/index.php?title=Wav_File_Sink#Source_Files) to the `\details` section
* Replace all occurences of "wav file header" in the comments with "audio file metadata" or similar
* Apply the above to `wavfile_sink.h` as well

### System Information

OS: –
GR Installation Method: –

### GNU Radio Version

3.11-git (main)

### Specific Version

3.9.0.0 and later

### Steps to Reproduce the Problem

reading the docstrings

drmpeg, 2022-06-27:
I wouldn't say _all_ of libsndfile. There are many other formats that are not being selected. Here's the complete list (scroll down to File Open Function):

https://libsndfile.github.io/libsndfile/api.html

Some formats that may be useful are Matlab/Octave and MP3 (which has just been added).

harshiiiit, 2023-01-30:
Hey, I would like to work on this issue as my first issue. I have figured out what changes to make to solve it, Thanks :)

marcusmueller, 2024-01-31:
yes, this remains outstanding and a good first issue, sadly Harshiit never came back.