gnuradio/gnuradio #5972: No audio in FM receiver

elucidator99, 2022-06-26:
I am beginner in SDR and I have issue with the ouput audio for FM receiver that i followed the graph from youtube tutorial (https://www.youtube.com/watch?v=WACYYbNXMCU&t=312s&ab_channel=MKhanfar)

i use ubuntu 22.04 LTS in Oracle VM VirtualBox 6.0
i use GNU Radio Companion 3.10.1. (python 3.10.4)
i use HackRF One

[image]
i tried with simple audio sink and the sound come out.

but when i tried with FM receiver, the sound not come out but the graph running
[image]
[image]

I have tried to change the device name but nothing happenend.
Please help
Thank you.

n1ai, 2022-06-26:
It may help if you attach your flowgraph to this report.  The github web page allows you to attach files by dragging and dropping them into the input text area.

marcusmueller, 2022-06-26:
Notice the aUaU and OOOO: this is an indication of your sample rates not fitting together, and a glimpse at your flowgraph would seem to confirm that!

So, that's not a GNU Radio bug, but a request for help! Sadly, this bug tracker is only for GNU Radio bugs; we also describe what goes wrong in our [sample rate tutorial](https://wiki.gnuradio.org/index.php?title=Sample_Rate_Tutorial)! If you haven't, do the GNU Radio tutorials on https://tutorials.gnuradio.org from start (in order!) to the FM transceiver tutorial. That's super rewarding, as it helps you understand the things that might go wrong!

If you already did that, and your questions remain unanswered: I think it'd be best if you signed up for the discuss-gnuradio@gnu.org mailing list [here](https://lists.gnu.org/mailman/listinfo/discuss-gnuradio) and then sent an email.

closed 2022-06-26 by marcusmueller

elucidator99, 2022-06-27:
I'm sorry for inconvenience and thank you for your advice.
I'll try to understand the tutorial and when the thing still remail i'll send email to discuss.

Have a nice day
Thank you