gnuradio/gnuradio #5973: i686 test failures

unspecd, 2022-06-28:
### What happened?

I am working on updating the GNU Radio from 3.8.0.0 to 3.10.3.0. I'm seeing failures when building on i686.

Here is the build log: https://github.com/void-linux/void-packages/runs/7086379225?check_suite_focus=true#step:7:10646

### System Information

OS: Void Linux (kernel 5.18.7, gcc 10.2.1, python 3.10.5, pybind 2.9.2, pygccxml 2.2.1)

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.3.0

### Steps to Reproduce the Problem

```
$ git clone https://github.com/void-linux/void-packages && cd void-packages
$ curl -L https://github.com/void-linux/void-packages/pull/35049.patch | git apply -
$ ./xbps-src binary-bootstrap
$ ./xbps-src -Q pkg gnuradio
```

### Relevant log output

```shell
[0/1] Running tests...
Test project /builddir/gnuradio-3.10.3.0/build
        Start   1: runtime_qa_buffer.cc
  1/261 Test   #1: runtime_qa_buffer.cc .........................   Passed    5.86 sec
        Start   2: runtime_qa_io_signature.cc
  2/261 Test   #2: runtime_qa_io_signature.cc ...................   Passed    0.01 sec
        Start   3: runtime_qa_logger.cc

... skipped ...

        Start  63: qa_moving_average
 63/261 Test  #63: qa_moving_average ............................***Failed    0.37 sec
...F.
======================================================================
FAIL: test_vector_complex (__main__.test_moving_average)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/builddir/gnuradio-3.10.3.0/gr-blocks/python/blocks/qa_moving_average.py", line 150, in test_vector_complex
    self.assertEqual(dut_data, ref_data)
AssertionError: Lists differ: [(705[443 chars].3553466796875+406.9476318359375j), (208.45861[1463 chars]25j)] != [(705[443 chars].35531616210938+406.9476318359375j), (208.4586[1474 chars]75j)]
First differing element 11:
(207.3553466796875+406.9476318359375j)
(207.35531616210938+406.9476318359375j)
Diff is 4376 characters long. Set self.maxDiff to None to see it.
----------------------------------------------------------------------
Ran 5 tests in 0.124s
FAILED (failures=1)
        Start  64: qa_multiply_conjugate
 64/261 Test  #64: qa_multiply_conjugate ........................   Passed    0.25 sec
        Start  65: qa_multiply_matrix_xx
 65/261 Test  #65: qa_multiply_matrix_xx ........................   Passed    0.27 sec
        Start  66: qa_mute
 66/261 Test  #66: qa_mute ......................................   Passed    0.25 sec

... skipped ...

        Start 125: qa_window
125/261 Test #125: qa_window ....................................***Failed    0.22 sec
F
======================================================================
FAIL: test_normwin (__main__.test_window)
Verify window normalization
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/builddir/gnuradio-3.10.3.0/gr-fft/python/fft/qa_window.py", line 38, in test_normwin
    self.assertAlmostEqual(power, 1.0)
AssertionError: 0.9999999411865508 != 1.0 within 7 places (5.8813449221872816e-08 difference)
----------------------------------------------------------------------
Ran 1 test in 0.001s
FAILED (failures=1)
        Start 126: filter_qa_firdes.cc
126/261 Test #126: filter_qa_firdes.cc ..........................   Passed    0.01 sec
        Start 127: filter_qa_fir_filter_with_buffer.cc
127/261 Test #127: filter_qa_fir_filter_with_buffer.cc ..........   Passed    0.18 sec
        Start 128: filter_qa_mmse_fir_interpolator_cc.cc

... skipped ...

        Start 156: qa_ctcss_squelch
156/261 Test #156: qa_ctcss_squelch .............................***Failed    0.27 sec
.FF
======================================================================
FAIL: test_ctcss_squelch_002 (__main__.test_ctcss_squelch)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/builddir/gnuradio-3.10.3.0/gr-analog/python/analog/qa_ctcss_squelch.py", line 70, in test_ctcss_squelch_002
    self.assertFloatTuplesAlmostEqual(expected_result, result_data, 4)
  File "/builddir/gnuradio-3.10.3.0/gnuradio-runtime/python/gnuradio/gr_unittest.py", line 96, in assertFloatTuplesAlmostEqual
    self.assertEqual(len(a), len(b))
AssertionError: 39 != 29
======================================================================
FAIL: test_ctcss_squelch_003 (__main__.test_ctcss_squelch)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/builddir/gnuradio-3.10.3.0/gr-analog/python/analog/qa_ctcss_squelch.py", line 95, in test_ctcss_squelch_003
    self.assertFloatTuplesAlmostEqual(expected_result, result_data, 4)
  File "/builddir/gnuradio-3.10.3.0/gnuradio-runtime/python/gnuradio/gr_unittest.py", line 97, in assertFloatTuplesAlmostEqual
    return all([
  File "/builddir/gnuradio-3.10.3.0/gnuradio-runtime/python/gnuradio/gr_unittest.py", line 98, in <listcomp>
    self.assertAlmostEqual(x, y, places, msg)
AssertionError: 0.8 != 0.0 within 4 places (0.8 difference)
----------------------------------------------------------------------
Ran 3 tests in 0.006s
FAILED (failures=2)
        Start 157: qa_dpll
157/261 Test #157: qa_dpll ......................................   Passed    0.27 sec
        Start 158: qa_fastnoise
158/261 Test #158: qa_fastnoise .................................   Passed   13.41 sec
        Start 159: qa_fm_emph

... skipped ...

        Start 180: qa_constellation
180/261 Test #180: qa_constellation .............................***Failed    9.62 sec
..FF.
======================================================================
FAIL: test_soft_qam16_calc (__main__.test_constellation)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/builddir/gnuradio-3.10.3.0/gr-digital/python/digital/qa_constellation.py", line 379, in test_soft_qam16_calc
    self.assertFloatTuplesAlmostEqual(y_cpp_raw_calc, y_cpp_table, 0)
  File "/builddir/gnuradio-3.10.3.0/gnuradio-runtime/python/gnuradio/gr_unittest.py", line 96, in assertFloatTuplesAlmostEqual
    self.assertEqual(len(a), len(b))
AssertionError: 17 != 25
======================================================================
FAIL: test_soft_qpsk_calc (__main__.test_constellation)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/builddir/gnuradio-3.10.3.0/gr-digital/python/digital/qa_constellation.py", line 331, in test_soft_qpsk_calc
    self.assertEqual(y_cpp_raw_calc, y_cpp_table)
AssertionError: Lists differ: [0.0,[16 chars]1.0, 1.0, 0.0, 1.0, 1.0, 0.0, 0.0, 1.0, 0.0, 1[13 chars] 1.0] != [0.0,[16 chars]1.0, 0.0, 1.0, 0.0, 1.0, 0.0, 1.0, 0.0, 0.0, 1[28 chars] 1.0]
First differing element 5:
1.0
0.0
Second list contains 3 additional elements.
First extra element 17:
0.0
  [0.0,
   0.0,
   1.0,
   0.0,
   1.0,
+  0.0,
   1.0,
+  0.0,
+  1.0,
+  0.0,
+  1.0,
+  0.0,
   0.0,
   1.0,
   1.0,
   0.0,
-  0.0,
   1.0,
   0.0,
-  1.0,
-  1.0,
   0.0,
   1.0]
----------------------------------------------------------------------
Ran 5 tests in 9.333s
FAILED (failures=2)
        Start 181: qa_constellation_decoder_cb
181/261 Test #181: qa_constellation_decoder_cb ..................   Passed    0.28 sec
        Start 182: qa_constellation_encoder_bc
182/261 Test #182: qa_constellation_encoder_bc ..................   Passed    0.32 sec
        Start 183: qa_constellation_receiver

... skipped ...

259/261 Test #259: qa_udp_sink ..................................   Passed    0.55 sec
        Start 260: qa_udp_source
260/261 Test #260: qa_udp_source ................................   Passed    0.55 sec
        Start 261: qa_soapy_types
261/261 Test #261: qa_soapy_types ...............................   Passed    0.25 sec
98% tests passed, 4 tests failed out of 261
Total Test time (real) = 170.02 sec
The following tests FAILED:
	 63 - qa_moving_average (Failed)
	125 - qa_window (Failed)
	156 - qa_ctcss_squelch (Failed)
	180 - qa_constellation (Failed)
Errors while running CTest
FAILED: CMakeFiles/test.util 
cd /builddir/gnuradio-3.10.3.0/build && /usr/bin/ctest --force-new-ctest-process
ninja: build stopped: subcommand failed.
```

marcusmueller, 2022-06-28:
The qa_window issue is strange to appear only on an i686 build: I can see how we introduce a small error, as the window::build method actually calculates a `double` error, but normalizes with a single-precision `float`:

https://github.com/gnuradio/gnuradio/blob/acf4658bfb379ac2cce7f57b257a13a35713c919/gr-fft/lib/window.cc#L386

Python only works in `double` all the way down. However, the FPUs of different x86 CPUs should have the same error here, no matter whether in i686 or x86_64 mode. We should be seeing this fail in our own unit testing all the time, or you should not be seeing it. Unlike your build, I get an error of 3.4983966168411484e-08 (you get 5.8813449221872816e-08), so my error correctly rounds down to less than 10⁻⁷. 

I'm a bit surprised our two FPUs are so different. @unspecd could you tell me what your CPU is, exactly?

---
I can comment a bit on the moving average failure:

Interesting; since at least e76d04ca2f4f15e3b1a1ab2a81dd52c4e6d2472c , the moving average math is done through VOLK in  the vector case, and in plain C++ in the single-element case. The unit test that's failing here compares these two.

looking at that, maybe @ThomasHabets has a comment on whether it'd make sense to unify on using VOLK for both cases, i.e. for both 

https://github.com/gnuradio/gnuradio/blob/acf4658bfb379ac2cce7f57b257a13a35713c919/gr-blocks/lib/moving_average_impl.cc#L182-L188

and for https://github.com/gnuradio/gnuradio/blob/acf4658bfb379ac2cce7f57b257a13a35713c919/gr-blocks/lib/moving_average_impl.cc#L192-L202

@jdemel is this perhaps worth opening as VOLK issue? Do we do i686 (i.e., 32 bit x86) testing?

----

the squelch and constellation mod errors are very strange to me, and I've not seen them before. 

The only explanation for the squelch failure is that really basic math or things like `std::cos` work differently on your machine than they've ever worked for us. And I see we're not great at testing on 32 bit machines, it seems, but these squelch blocks have been around practically forever! Really confused by what's happening here.

@unspecd: does the void build infra use any specific compiler flags I'm not aware of?

marcusmueller, 2022-06-29:
@jdemel  ah, thanks! On one hand, a bit sad, on the other, I'm having a hard time finding a 32bit x86 system these days; hm.

marcusmueller, 2022-06-29:
@unspecd hey, I've been digging a tiny bit, and debian seems to build things well on 32 bit: https://buildd.debian.org/status/fetch.php?pkg=gnuradio&arch=i386&ver=3.10.3.0%7Erc1-1&stamp=1656071459&raw=0

bit stumped on how to reproduce; will probably need your input on how things might be different.

argilo, 2022-06-30:
`test_ctcss_squelch_002` and `test_ctcss_squelch_003` look shoddy to me. The input signal doesn't appear to have any connection to CTCSS, so I think the tests are just asserting that the block's behaviour when given a garbage input remains the same as when the test was originally written.

argilo, 2022-06-30:
The test is also passing in some ridiculous parameters though, like a sample rate of 1 sps!

unspecd, 2022-06-30:
Hi all, thanks for the efforts in solving this issue.
I will share more details on the weekend (I'm a bit busy at the moment).

rear1019, 2022-07-09:
> However, the FPUs of different x86 CPUs should have the same error here, no matter whether in i686 or x86_64 mode

Assuming the same FPU is used. This is not true by default in GCC. See `-mfpmath` flag.

argilo, 2022-10-15:
I opened pull requests to fix three of the failing tests:

* `qa_moving_average`: https://github.com/gnuradio/gnuradio/pull/6246
* `qa_window`: https://github.com/gnuradio/gnuradio/pull/6247
* `qa_constellation`: https://github.com/gnuradio/gnuradio/pull/6248

`qa_ctcss_squelch` might take a bit more work, since the failing tests look shoddy and may need to be rewritten.

On my Debian 11 test system, some more tests are failing as well:

* `qa_polar_encoder_systematic` and `qa_polar_decoder_sc_systematic` crash due to a bug in Debian's version of libvolk: https://github.com/gnuradio/gnuradio/issues/5013#issuecomment-1279645954
* `qa_fecapi_ldpc` fails, specifically this test:

https://github.com/gnuradio/gnuradio/blob/09930fdcc5c892eaff5754f6ef0203146bc1cca7/gr-fec/python/fec/qa_fecapi_ldpc.py#L108-L122

I'm not sure yet whether the test or the functional code is broken there.

argilo, 2022-10-15:
* `qa_fecapi_ldpc`: #6254

argilo, 2022-10-16:
* `qa_ctcss_squelch`: #6257

All i686 test failures should now have open pull requests.

argilo, 2022-10-26:
All test failures have now been fixed on the `main` branch, and I verified that no more test failures occur. :tada:

Since this bug report pertained to `maint-3.10`, I'll leave it open until the fixes are backported to that branch.

argilo, 2022-11-18:
The last of the i686 bugs will be fixed on the `maint-3.10` branch as soon as #6385 is merged.

argilo, 2022-11-18:
i686 is also now included in CI, which should help avoid new bugs.

closed 2022-11-18 by argilo