gnuradio/gnuradio #5978: GRC segfault on start-up

duggabe, 2022-06-30:
### What happened?

When starting gnuradio-companion from a terminal screen, the GRC screen opens briefly and then closes. This happens sporatically and does not depend on any specific flowgraph. It even happens after deleting ~/.gnuradio/grc.conf

### System Information

OS: Ubuntu 22.04
GR Installation Method: from source. GNU Radio Companion 3.10.2.0
GNOME Version 42.2
Windowing System: Wayland
Block paths:
	/usr/local/share/gnuradio/grc/blocks
I have cmake and make logs from the build if needed.

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.2.0

### Steps to Reproduce the Problem

Open a terminal screen.
Enter gnuradio-companion

### Relevant log output

```shell
Jun 29 07:18:29 LENOVO wpa_supplicant[865]: wlp9s0: CTRL-EVENT-SIGNAL-CHANGE above=1 signal=-13 noise=9999 txrate=866700
Jun 29 07:18:29 LENOVO gnome-shell[1416]: Can't update stage views actor <dashtodockContainer>[<Gjs_ubuntu-dock_ubuntu_com_docking_DashToDock>:0x5593ae792b50] is on because it needs >
Jun 29 07:18:29 LENOVO gnome-shell[1416]: Can't update stage views actor <unnamed>[<Gjs_ubuntu-dock_ubuntu_com_docking_DashSlideContainer>:0x5593aa955d80] is on because it needs an a>
Jun 29 07:18:29 LENOVO gnome-shell[1416]: Can't update stage views actor <dashtodockBox>[<StBoxLayout>:0x5593ae6fc100] is on because it needs an allocation.
Jun 29 07:18:29 LENOVO gnome-shell[1416]: Can't update stage views actor <dash>[<Gjs_ubuntu-dock_ubuntu_com_dash_DockDash>:0x5593ae755870] is on because it needs an allocation.
Jun 29 07:18:29 LENOVO gnome-shell[1416]: Can't update stage views actor <dashtodockDashContainer>[<StBoxLayout>:0x5593ab584c50] is on because it needs an allocation.
Jun 29 07:18:29 LENOVO gnome-shell[1416]: Can't update stage views actor <dashtodockDashScrollview>[<StScrollView>:0x5593ae7624f0] is on because it needs an allocation.
Jun 29 07:18:29 LENOVO gnome-shell[1416]: Can't update stage views actor <unnamed>[<StBoxLayout>:0x5593ae745520] is on because it needs an allocation.
Jun 29 07:18:29 LENOVO gnome-shell[1416]: Can't update stage views actor <dashtodockContainer>[<Gjs_ubuntu-dock_ubuntu_com_docking_DashToDock>:0x5593ae792b50] is on because it needs >
Jun 29 07:18:29 LENOVO gnome-shell[1416]: Can't update stage views actor <unnamed>[<Gjs_ubuntu-dock_ubuntu_com_docking_DashSlideContainer>:0x5593aa955d80] is on because it needs an a>
Jun 29 07:18:29 LENOVO gnome-shell[1416]: Can't update stage views actor <dashtodockBox>[<StBoxLayout>:0x5593ae6fc100] is on because it needs an allocation.
Jun 29 07:18:29 LENOVO gnome-shell[1416]: Can't update stage views actor <dash>[<Gjs_ubuntu-dock_ubuntu_com_dash_DockDash>:0x5593ae755870] is on because it needs an allocation.
Jun 29 07:18:29 LENOVO gnome-shell[1416]: Can't update stage views actor <dashtodockDashContainer>[<StBoxLayout>:0x5593ab584c50] is on because it needs an allocation.
Jun 29 07:18:29 LENOVO gnome-shell[1416]: Can't update stage views actor <dashtodockDashScrollview>[<StScrollView>:0x5593ae7624f0] is on because it needs an allocation.
Jun 29 07:18:29 LENOVO gnome-shell[1416]: Can't update stage views actor <unnamed>[<StBoxLayout>:0x5593ae745520] is on because it needs an allocation.
Jun 29 07:18:29 LENOVO kernel: show_signal_msg: 6 callbacks suppressed
Jun 29 07:18:29 LENOVO kernel: gnuradio-compan[11440]: segfault at 28 ip 00007f5471997c3c sp 00007f5468e8ee70 error 4 in libglib-2.0.so.0.7200.1[7f5471961000+8f000]
Jun 29 07:18:29 LENOVO kernel: Code: 48 83 ec 08 8b 47 18 85 c0 0f 8e 9f 00 00 00 4c 8b 6b 20 48 89 f5 41 89 d4 4d 85 ed 74 08 4c 89 ef e8 18 05 05 00 48 8b 43 58 <8b> 40 28 85 c0 74>
```

drmpeg, 2022-07-02:
I'm also seeing this on Ubuntu 22.04 after an update to libglib2.0. This isn't a GNU Radio issue, it's an Ubuntu issue. A bug report needs to be filed with Ubuntu, preferably with a GDB trace.

duggabe, 2022-07-02:
@drmpeg how do I set up the GDB trace? I'm not familiar with that.

drmpeg, 2022-07-02:
https://wiki.gnuradio.org/index.php/TutorialsDebugging

Specifically, compile GNU Radio with ```-DCMAKE_BUILD_TYPE=Debug``` and enable core files with ```ulimit -c unlimited```.

drmpeg, 2022-07-02:
BTW, libglib2.0 wasn't updated. I got fooled when I saw it in the dpkg.log file because it gets triggered on many updates. It's still the original version that shipped with Ubuntu 22.04.

duggabe, 2023-09-22:
This has not occurred with Ubuntu 22.04.2. No longer an issue.

closed 2023-09-22 by duggabe