gnuradio/gnuradio #5985: "Trellis Metrics" in-tree block produces traceback due to missing attributes

n1ai, 2022-07-02:
### What happened?

I am trying to port a flowgraph that works on GR 3.8:

https://github.com/M17-Project/gr-m17/blob/main/examples/m17_viterbi.grc

Note that it needs the m17.fsm file in order to work:

https://github.com/M17-Project/gr-m17/blob/main/examples/m17.fsm

It includes the Trellis Metrics block:

https://wiki.gnuradio.org/index.php/Trellis_Metrics

That block's YAML is:

https://github.com/gnuradio/gnuradio/blob/master/gr-trellis/grc/trellis_metrics_x.block.yml

The YAML file has:

templates:
    imports: from gnuradio import trellis, digital
    make: trellis.metrics_${type}(${card}, ${dim}, ${table}, ${metric_type})

So it expects to use a trellis.metrics_${type}(${card} object.

To confirm, the generated python for the flowgraph contains the line:

self.trellis_metrics_x_0 = trellis.metrics_f(fsm.O(), fsm.I(), [-1, -1, -1, 1, 1, -1, 1, 1], digital.TRELLIS_EUCLIDEAN)

Running the flowgraph produces:

Traceback (most recent call last):
  File "/home/dcherkus/code/sdr/gr-m17.M17-Project/examples/m17_viterbi.py", line 325, in <module>
    main()
  File "/home/dcherkus/code/sdr/gr-m17.M17-Project/examples/m17_viterbi.py", line 303, in main
    tb = top_block_cls(frame_size=options.frame_size)
  File "/home/dcherkus/code/sdr/gr-m17.M17-Project/examples/m17_viterbi.py", line 97, in __init__
    self.trellis_metrics_x_0 = trellis.metrics_f(fsm.O(), fsm.I(), [-1, -1, -1, 1, 1, -1, 1, 1], digital.TRELLIS_EUCLIDEAN)
AttributeError: module 'gnuradio.trellis' has no attribute 'metrics_f'

I have looked at the python code in the interpreter and the GR 3.10 version 
of trellis objects has no "metrics_f" attributes nor any starting with 
"metrics".

Looking at the python code in the interpreter in a GR 3.8 environment I see:

$ python3
Python 3.6.9 (default, Mar 15 2022, 13:55:28) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from PyQt5 import Qt
>>> from gnuradio import qtgui
>>> from gnuradio.filter import firdes
>>> import sip
>>> from gnuradio import blocks
>>> from gnuradio import digital
>>> from gnuradio import fec
>>> from gnuradio import gr
>>> from gnuradio.fft import window
>>> import sys
>>> import signal
>>> from argparse import ArgumentParser
>>> from gnuradio.eng_arg import eng_float, intx
>>> from gnuradio import eng_notation
>>> from gnuradio import trellis
>>> from gnuradio import trellis, digital
>>> help(trellis.metrics_f)
Help on function make in module gnuradio.trellis.trellis_swig2:

make(O:'int', D:'int', TABLE:'pmt_vector_float', TYPE:'gr::digital::trellis_metric_type_t') -> 'gr::trellis::metrics< float >::sptr'
    make(int O, int D, pmt_vector_float TABLE, gr::digital::trellis_metric_type_t TYPE) -> metrics_f_sptr
    
    Evaluate metrics for use by the Viterbi algorithm.
    
    Constructor Specific Documentation:
    
    Args:
        O : 
        D : 
        TABLE : 
        TYPE :

>>> [ x for x in dir(trellis) if x.startswith("metric") ]
['metrics_c', 'metrics_c_make', 'metrics_c_sptr',
 'metrics_c_sptr_swigregister', 'metrics_c_swigregister', 'metrics_f',
 'metrics_f_make', 'metrics_f_sptr', 'metrics_f_sptr_swigregister',
 'metrics_f_swigregister', 'metrics_i', 'metrics_i_make', 'metrics_i_sptr',
 'metrics_i_sptr_swigregister', 'metrics_i_swigregister', 'metrics_s',
 'metrics_s_make', 'metrics_s_sptr', 'metrics_s_sptr_swigregister',
 'metrics_s_swigregister']

So, the metrics attributes are present in GR 3.8 and not in GR 3.10, yet
an in-tree block expects them to be present.

I've looked at the porting guides:
- https://wiki.gnuradio.org/index.php/GNU_Radio_3.9_OOT_Module_Porting_Guide
- https://wiki.gnuradio.org/index.php/GNU_Radio_3.10_OOT_Module_Porting_Guide

And there is no mention of changes to how trellis works.

And, note, I am not working with an OOT, "Trellis Metrics" is an in-tree block.

If this block is not going to be fixed, I would be happy if a subject-matter expert would show me how to make this work on GR 3.10, since it was working on GR 3.8, and I'm having a hard time figuring out all the "moving parts" in this space.

I believe this is a different issue than https://github.com/gnuradio/gnuradio/issues/4834 since that one is about C++ bindings and that one did not work on GR 3.8 whereas mine does.

### System Information

OS: Linux
GR Installation method: Conda for GR 3.10.2, build from source for GR 3.8.

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.2.0

### Steps to Reproduce the Problem

I believe any flowgraph using Trellis Metrics in GR 3.10 fails.

Otherwise, see problem statement.

### Relevant log output

```shell
Traceback (most recent call last):
  File "/home/dcherkus/code/sdr/gr-m17.M17-Project/examples/m17_viterbi.py", line 325, in <module>
    main()
  File "/home/dcherkus/code/sdr/gr-m17.M17-Project/examples/m17_viterbi.py", line 303, in main
    tb = top_block_cls(frame_size=options.frame_size)
  File "/home/dcherkus/code/sdr/gr-m17.M17-Project/examples/m17_viterbi.py", line 97, in __init__
    self.trellis_metrics_x_0 = trellis.metrics_f(fsm.O(), fsm.I(), [-1, -1, -1, 1, 1, -1, 1, 1], digital.TRELLIS_EUCLIDEAN)
AttributeError: module 'gnuradio.trellis' has no attribute 'metrics_f'
```

n1ai, 2022-07-04:
Thanks for the analysis, @ryanvolz !

Any suggestion for the next steps?

I see from similarity to issue https://github.com/gnuradio/gnuradio/issues/5612 that labels such as "Bug" and "pybind11" might be appropriate?

Does someone else come along and apply these labels?

ryanvolz, 2022-07-04:
Anyone who understands pybind11 could put in a PR to add the bindings. It might be possible to do it by analogy with other existing bindings, for anyone like me who doesn't really know pybind11 that well.

n1ai, 2022-07-04:
I'm also not up to speed on pybind11, or on trellis either.  

Hopefully the analysis and the labels will raise the visibility of the issue.

mcdiarmid, 2022-07-14:
When I posted #4834 there were bindings missing from a majority of the gr-trellis module.  This was addressed in #5707, but [metrics](https://github.com/gnuradio/gnuradio/blob/main/gr-trellis/python/trellis/bindings/metrics_python.cc) and [quicksort_index](https://github.com/gnuradio/gnuradio/blob/main/gr-trellis/python/trellis/bindings/quicksort_index_python.cc) appear to have been missed

PR #6022 (2022-07-25): gr-trellis: add correct python bindings for metrics

# Pull Request Details

The metrics_python.cc file does not contain the correct bindings.
The missing bindings where added manually.

## Related Issue

Fixes #5985

## Testing Done

Tested with example from #5985
## Checklist

- [x] I have read the
commit 76d8fb1d (2022-07-25): gr-trellis: add correct python bindings for metrics

Fixes #5985
PR #6093 (2022-08-14): gr-trellis: add correct python bindings for metrics (backport to maint-3.10)
Fixes #5985

(cherry picked from commit 76d8fb1d2858a78ba0a949079d19908a326eef8e)

Backport https://github.com/gnuradio/gnuradio/pull/6022
commit ccb41214 (2022-08-14): gr-trellis: add correct python bindings for metrics

Fixes #5985

(cherry picked from commit 76d8fb1d2858a78ba0a949079d19908a326eef8e)
closed 2022-07-25 by mormj