gnuradio/gnuradio #6000: QT Frequency Sink does not update

JayKickliter, 2022-07-07:
### What happened?

QT Frequency sink renders only once then never updates.

### System Information

OS: macOS 12.2.1 Monterey, Apple M1 Pro
GR Installation Method: Both Homebrew and from-source build show the same behavior.

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.1.0

### Steps to Reproduce the Problem

Create a flow graph of a [random noise source] -> [throttle] -> [QT Frequency Sink]

JayKickliter, 2022-07-13:
Following up, it's not just the freq sink. All QT GUI sinks I've tried so far freeze after a single render.

mormj, 2022-08-19:
Could be fixed by recent commit: https://github.com/gnuradio/gnuradio/commit/3db403b3dcf272eebd85ac2c15934fa3ade12abd

Are you able to try the latest maint-3.10?

JayKickliter, 2022-08-19:
> Are you able to try the latest maint-3.10?

yep, that did it.

Fixed by https://github.com/gnuradio/gnuradio/commit/3db403b3dcf272eebd85ac2c15934fa3ade12abd

JayKickliter, 2022-08-19:
Fixed by https://github.com/gnuradio/gnuradio/commit/3db403b3dcf272eebd85ac2c15934fa3ade12abd

JayKickliter, 2022-08-19:
Fixes

closed 2022-08-19 by JayKickliter