gnuradio/gnuradio #6003: Invert on tag

michael-roe, 2022-07-10:
### Feature Description

This would be a block that has a stream of floats as input and output, and is parameterized by two tags, negTag and posTag.

When the block sees negTag on a sample, it starts inverting samples until it sees posTag on a sample.

The use case is Binary Phase Shift Keying, where there is an ambiguity in the phase. The idea is that previous blocks will have detected the phase somehow (e.g. looking for  sync words) and have put in tags where the phase is detected to have changed.

### Feature Urgency

medium (would be nice to have in the near future)

willcode, 2022-07-10:
There is a Multiply By Tag Value block. Is that sufficient?

michael-roe, 2022-07-10:
I was going to say it wasn't sufficient, because I want to have two correlate access code tag blocks marking the sync word and the negation of the sync word, and then use that to invert the sense.

If correlate access code tag could add a tag with a value of -1, that might do it.