gnuradio/gnuradio #6006: Building 'rstest.exe'

gvanem, 2022-07-11:
### Feature Description

I was trying to build a working `rstest.exe` on Win-10 using MSVC-2022 (or clang-cl) with support for `-DALL_VERSIONS`.
But have several issues with it.
* First there is no `init_rs_int()` function.
* Several function should be exported as `"C"` (but are not).
* `reed-solomon/init_rs.c` does not support compiling with `-DCCSDS` (no element `alpha_to`).

### Feature Urgency

low (just an idea)

### More Information

I guess a `reed-solon/int.c` is needed with only this:
```c
#include "int.h"
#include "decode_rs.h"
#include "encode_rs.h"
```

And also have the `reed-solomon/init_rs.c` be compiled twice; with an extra `-DBIGSYM` added.

drmpeg, 2022-07-11:
Perhaps it would be better to work with the original code?

https://github.com/quiet/libfec

http://www.ka9q.net/code/fec/

gvanem, 2022-07-11:
> Perhaps it would be better to work with the original code?

Or just clean-up and remove rstest.c from GnuRadio? It's rather messy and confusing now IMHO.