gnuradio/gnuradio #6008: Build failure in freedv with codec2 1.0.4

bulvara, 2022-07-13:
### What happened?

```
./gnuradio/gr-vocoder/lib/freedv_rx_ss_impl.cc: In constructor 'gr::vocoder::freedv_rx_ss_impl::freedv_rx_ss_impl(int, float, int)':
./gnuradio/gr-vocoder/lib/freedv_rx_ss_impl.cc:41:15: error: 'struct freedv_advanced' has no member named 'interleave_frames'
   41 |         d_adv.interleave_frames = interleave_frames;
      |               ^~~~~~~~~~~~~~~~~
[ 33%] Built target gnuradio-blocks
[ 33%] Building CXX object gr-uhd/examples/c++/CMakeFiles/tags_demo.dir/tags_demo.cc.o
./gnuradio/gr-vocoder/lib/freedv_tx_ss_impl.cc: In constructor 'gr::vocoder::freedv_tx_ss_impl::freedv_tx_ss_impl(int, std::string, int)':
./gnuradio/gr-vocoder/lib/freedv_tx_ss_impl.cc:39:15: error: 'struct freedv_advanced' has no member named 'interleave_frames'
   39 |         d_adv.interleave_frames = interleave_frames;
      |               ^~~~~~~~~~~~~~~~~
make[2]: *** [gr-vocoder/lib/CMakeFiles/gnuradio-vocoder.dir/build.make:365: gr-vocoder/lib/CMakeFiles/gnuradio-vocoder.dir/freedv_tx_ss_impl.cc.o] Error 1
make[2]: *** [gr-vocoder/lib/CMakeFiles/gnuradio-vocoder.dir/build.make:349: gr-vocoder/lib/CMakeFiles/gnuradio-vocoder.dir/freedv_rx_ss_impl.cc.o] Error 1
```

### System Information

ArchLinux

### GNU Radio Version

3.11-git (main)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

Build from source

haakov, 2022-07-14:
I've got the same issue (same system), downgrading to 1.0.3 works. [The issue appears to be fixed in 1.0.4 rc2](https://github.com/drowe67/codec2/releases/tag/v1.0.4_rc2), released less than 24 hours ago.

[The Arch package](https://archlinux.org/packages/community/x86_64/codec2/) has already been flagged out of date, so I suppose this problem will solve itself in a day or two.

haakov, 2022-07-24:
@bulvara The package has been updated to 1.0.5. Can you confirm that the issue has been fixed?

marcusmueller, 2024-01-31:
seems to be solved.

closed 2024-01-31 by marcusmueller