gnuradio/gnuradio #6010: Duplicate "All files" filter entry in save screenshot dialog

haakov, 2022-07-14:
### What happened?

I was saving several screenshots of a flowgraph using the Screen Capture (Ctrl + P) function. After saving a few screenshots I noticed the filter dropdown menu looking like this: 
[image]

I had expected to see only a single "All files" entry. 

### System Information

OS: Arch Linux
GR Installation Method: Source

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.11.0.0git-185-g53dd57ca

### Steps to Reproduce the Problem

1. Open a flowgraph
2. Hit Ctrl+P, then Esc.
3. Repeat step 2 a couple times
4. Hit Ctrl+P, then open the filter dropdown menu ("PDF Files")

harxish, 2022-07-17:
@haakov Is this still open? Can I work on this?

haakov, 2022-07-17:
Yes, certainly!

harxish, 2022-07-23:
Hey, could you just say me how I run this ? When I run the main.py file from grc it says me `ValueError: Namespace Gtk is already loaded with version 4.0`.
Is there something which I need to build before running this ?

harxish, 2022-07-23:
Hey, so I'm trying to run this application, but I'm not sure how....
I'm following this link https://wiki.gnuradio.org/index.php?title=CondaInstall#Building_GNU_Radio_from_source_within_a_conda_environment

But I'm not sure what build dir should I run the cmake from ??

Or is there anyway I can run main.py just like that ?

eat4toast, 2022-07-24:
> Hey, so I'm trying to run this application, but I'm not sure how.... I'm following this link [...]

For the first Question:
```
git clone https://github.com/gnuradio/gnuradio.git --depth=1
cd gnuradio && mkdir build
! run cmake here in build dir like(if you use Linux): 
   cmake -G "Ninja" -DCMAKE_PREFIX_PATH=$CONDA_PREFIX -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX -DLIB_SUFFIX="" ..

```
PS: using `cmake` to `build and install` will take some time(in my i5-11 takes about 25 minutes).

For the Question 2:
Can you clarify what means `main.py` here?
the conda environment will download the needed package and then we can build directly. If we did not choose conda but through the source, thus we may need to compile some extra packages manually.

harxish, 2022-07-24:
So what I mean by running main.py is:

When I make a change in some file in the grc/ dir and want to check if it works good, do I need to build everything again? Or is there anyway can see those changes instantly. 

Sorry for asking a lot of questions. I'm new to working with a project this big. But for the bug I think I can fix it.

eat4toast, 2022-07-24:
> So what I mean by running main.py is:
>
> When I make a change in some file in the grc/ dir and want to check if it works good, do I need to build everything again? Or is there anyway can see those [...]

thanks for your quick reply!  sorry，it's my time to going sleep😴. I am also a new one in this community，so i have no experience with your case. 

But from prior，i guess there is no need to re-build if just modify python related file. I will check tomorrow.

If you want to get the instant message（like discord，slack，telegram），consider using chatroom Matrix: [#gnuradio:gnuradio.org](https://chat.gnuradio.org/).

Also，the mailing list is a good place to post your question😃

eat4toast, 2022-07-24:
> So what I mean by running main.py is:
>
> When I make a change in some file in the grc/ dir and want to check if it works good, do I need to build everything again? Or is there anyway can see those [...]

Hi, harxish!
I think what this file might be you are looking for: https://github.com/gnuradio/gnuradio/blob/main/grc/main.py
in this file, it will call the GUI module and other modules under the directory.

haakov, 2022-07-25:
@harxish You can run GRC with `python -m grc` from the project root :)

rumbleFTW, 2022-10-14:
Greetings. May I work on this?

PR #6479 (2023-02-01): grc: fix duplicate All Files filter entry in save screenshot dialog

## Description
Fixes invoking screen capture (crtl+p) multiple times duplicates "All Files" option in the file filter each time it is invoked.

## Related Issue
Fixes #6010 

## Which blocks/areas does this affect?
grc/gui/FileDialogs.py line no. 60-63

## Testing Done
Yes
closed 2023-02-01 by willcode