gnuradio/gnuradio #6011: ImportError and missing libraries on Solus Linux

alexxtasi, 2022-07-18:
### What happened?

Just installed `gnuradio` from disto's packages and when tried to run it, `ImportError` windows appears.

### System Information

OS: Solus Linux
GR Installation Method: native distro packages (`eopkg`)

### GNU Radio Version

3.9  (maint-3.9)

### Specific Version

3.9.0.0-7 (viewed from distro's Software Center... the suggested command comes with an error)

### Steps to Reproduce the Problem

- Just install `gnuradio` on Solus with `eopkg` and try to run the application.

- I followed [ModuleNotFoundError](https://wiki.gnuradio.org/index.php/ModuleNotFoundError) wiki guide to solve the problem, but I cannot even determine the GNU Radio install prefix, because when running `gnuradio-config-info --prefix` the output is : 
```
gnuradio-config-info: error while loading shared libraries: libvolk.so.2.4: cannot open shared object file: No such file or directory
```

- Running `gnuradio-companion` this window appears : 
[image]

### Relevant log output

```shell
$ gnuradio-config-info

gnuradio-config-info: error while loading shared libraries: libvolk.so.2.4: cannot open shared object file: No such file or directory
```

velichkov, 2022-07-18:
Hi @alexxtasi,

> ImportError and missing libraries on Solus Linux
> 
> Just installed gnuradio from disto's packages and when tried to run it, ImportError windows appears.

This sound like packaging problem. My advice is to report this to the [package maintainer / Solus's development portal ](https://dev.getsol.us/)

> gnuradio-config-info: error while loading shared libraries: libvolk.so.2.4: cannot open shared object file: No such file or directory
>
> $ ls -l /usr/lib64/libvolk.so.2.5
> lrwxrwxrwx 1 root root 16 Ιουλ 18 10:09 /usr/lib64/libvolk.so.2.5 -> libvolk.so.2.5.1

This gnuradio-config-info was built for a different libvolk.so then the one you are having (`2.4` vs `2.5`). You need to uninstall gnuradio and rebuild it with the current volk version. I'm not familiar with Solus packages (`eopkg`) so I can't help you with this.

marcusmueller, 2022-07-23:
Fully agreeing with @velichkov: if you've somehow installed a different version of volk than solus ships, you need to undo that and re-install the solus version of GNU Radio and VOLK.

If the solus GNU Radio package does not depend on VOLK, that's a packaging bug that we sadly cannot do anything about; I honestly don't think we have a point of contact in the Solus community! (But: we'd be happy to have one, so if you talk to the solus package maintainer, tell her/him we'd love to hear from them!)

closed 2022-07-23 by marcusmueller