gnuradio/gnuradio #6016: Golay code

michael-roe, 2022-07-21:
### Feature Description

Lots of radio protocols use the Extended Binary Golay Code. There's an out-of-tree implementation on my github.

How would people feel about merging this into the main gnuradio codebase?

### Feature Urgency

medium (would be nice to have in the near future)

### More Information

See
https://github.com/michael-roe/gr-golay 

for an implementation, with tests.

dkozel, 2022-07-22:
Hi Michael,

I think it is reasonable to add a pair of Golay blocks to the FEC options available. @daniestevez has a Golay G24 code implementation in gr-satellites and he's looked at FEC in GNU Radio core recently if I remember correctly so drawing him into the discussion.
https://github.com/daniestevez/gr-satellites/tree/main/lib

You've already VOLK accelerated it which is good. There's some style comments I would make before merging to make the code, but the higher level question I'd like to ask (I don't have an answer) is if the blocks should actually be written and provided as FECAPI decoders and encoders. 

https://github.com/gnuradio/gnuradio/blob/main/gr-fec/include/gnuradio/fec/generic_decoder.h
https://github.com/gnuradio/gnuradio/blob/main/gr-fec/include/gnuradio/fec/generic_encoder.h

https://github.com/gnuradio/gnuradio/blob/main/gr-fec/lib/dummy_encoder_impl.cc
https://github.com/gnuradio/gnuradio/blob/main/gr-fec/lib/dummy_decoder_impl.cc

Finally, for completeness @mobilinkd has implementations of both the extended and perfect binary Golay codes in their M17 en/decoder. It would be nice to have both code versions available if possible, though certainly not a condition for merging.
https://github.com/mobilinkd/m17-cxx-demod/blob/master/include/m17cxx/Golay24.h

michael-roe, 2022-07-22:
Hi Derek,

Thanks for the reply.

A possible reason why not to use FECAPI is that I'm adding a tag to indicate that the block wasn't correctable. I think this goes beyond the API specified by FECAPI.

What do people think we should do here?

Mike

daniestevez, 2022-07-22:
Hi,

I think it would be useful to have a Golay code implementation in-tree. Some thoughts:

* I think it would be good to have `encode` and `decode` functions that work on single Golay codewords exposed in the API. The advantage of doing this is that someone else could use them in contexts where a streaming approach (as in your gr-golay blocks) is not applicable. For instance, some protocols have parts of the header that are Golay encoded, but the rest of the header or payload is not. The same functions can be called in the `work()` function of the streaming blocks.
* I wonder what would be the most useful input and output format for the streaming blocks. In gr-golay these are interpolate x2 and decimate x2 blocks that use a stream of bytes (containing unpacked bits) as input and output. There are other alternatives: Vectors of bytes of size 12 or 24 (containing unpacked bits) as input or output (this is how the in-tree ATSC and DVB-T Reed-Solomon blocks work); Having the data packed into shorts (for the 12 bits messages) or ints (for the 24 bits messages), which is less wasteful.
* Regarding what happens when the block isn't correctable, I guess that the options are either not generating an output (in which case the block is no longer a x2 decimator, and filtering output might be inconvenient in some applications) or marking the output in some way. Adding tags is a good way to mark the output as "wrong", but I hear that tags are not very efficient, so if the block is fed random input at high data rate, it might generate many output tags, killing performance. I guess that having an option to disable the generation of these tags would be nice. I wonder what do other similar in-tree block, such as the DTV Reed-Solomon blocks, do in this situation.

I'm not very familiar with FECAPI, but perhaps adjusting to this API gives answers to some of my questions, such as input/output formats.

daniestevez, 2022-07-22:
Something else: it's probably good to think about Golay(23, 12) as well, since this is very similar to Golay(24, 12). Regarding encoding/decoding, I have [some notes](https://destevez.net/2018/05/using-a-golay2412-decoder-for-golay2312/) about how the two codes are related. Essentially, if you have a Golay(24, 12) encoder/decoder, you can get a Golay(23, 12) encoder/decoder with very little work.

However, I'm more interested about input/output formats for streaming blocks that do Golay(23, 12). I guess that if we follow the same approach as in gr-golay, these blocks would interpolate/decimate with rates 23/12 and 12/23, which probably works OK.

michael-roe, 2022-07-22:
If you guys are in the mood for doing a small amount of code review, please could you also take a look at markov_source in https://github.com/michael-roe/gr-c4fm/tree/master/lib

It's related to this issue. markov_source implements a published NASA noise model, and the way you'ld typically use it is as a source of bit errors feeding into a error correction block like gr-golay.

daniestevez, 2022-07-22:
Can you share a link to the NASA model? I guess that the code will be easier to follow with a document at hand.

michael-roe, 2022-07-22:
https://ntrs.nasa.gov/api/citations/19810024821/downloads/19810024821.pdf
R.L. Miller, L. J. Deutsch, S.A. Butman, On the Error Statistics of Viterbi Decoding and the Performance of Concatenated Codes. NASA Jet Propulsion Laboratory, 1981.

daniestevez, 2022-07-30:
Hi @michael-roe, I've gone through the JPL report and your code.

I'm not sure of what your markov_source intends to do exactly. What the code suggests to me fits the following description:

*There is a Markov chain with states S0 = "no error", S1 = "error burst", and the corresponding state transition probabilities. For each output bit to generate, we advance the state of the Markov chain. Then, if we are in state S0, the output is never an error. If we are in state S1, the output is either an error or no error with probability 0.5.*

However, the code doesn't do exactly this. In particular, [L82](https://github.com/michael-roe/gr-c4fm/blob/master/lib/markov_source_b_impl.cc#L82) might take you out of S1 immediately after entering it. Also, whenever we enter or exit S1, the output is always an error (see [L59](https://github.com/michael-roe/gr-c4fm/blob/master/lib/markov_source_b_impl.cc#L59) and [L80](https://github.com/michael-roe/gr-c4fm/blob/master/lib/markov_source_b_impl.cc#L80)). I find these choices a bit weird, but perhaps I'm just missing the motivation.

Also, I wonder how this Markov chain relates to the JPL report. In the report, they model the burst errors at the output of a Viterbi decoder. The constraint length of the convolutional code shows up explicitly in the calculations. They consider the burst length and wait time between bursts. They show that these two quantities follow some modified geometric distributions.

To me, the Markov chain model for burst errors is quite different (and simpler) than the model in the report. For instance, in the Markov chain model the waiting time is "almost" distributed as a geometric distribution: P(W = s) = p_start*(1-p_start)^s. I say "almost" because in the way I introduced Markov chain there is always the chance that the state S1 is entered but no errors are produced in the output, so I wouldn't count this entering of S1 as a burst and hence we would have to sum up the waiting times to enter S1 this time and next time.

michael-roe, 2022-07-30:
Ok, so what I'm trying to do with this block is to approximate the burst errors at the output of a Viterbi decoder. As
 per the NASA JPL report (and confirmed by some experimentation) bursts are approximately geometrically distributed; and bursts of length 1 are possible.

The way I have set up the Markov chain in the model, entering S1 (or exiting) S1 always causes an error bit to be output, so you don't have "invisble" bursts with no errors. My motivation for doing it this way is it makes it easy to (a) observe an actual Viterbi decoder, and measure burst frequency and length; (b) plug these parameters into markov_source to get a noise source with the same(ish) statistical properties.

What I'm hearing in this discussion is that - at the very least - I ought to write documentation that better explains the use case and motivation.

But - if people think I should - I could change the model itself, so that e.g. the probablity of an error when first entering the burst state is 0.5 and so invisible bursts with no errors are possible.

michael-roe, 2022-07-30:
Also: bernoulli_source is straighforward, I think. The model that bit errors are independent with probability p is a useful one to have.

michael-roe, 2022-07-31:
The simpler Markov model - enter state S1 with probability p_start, leave S1 with probability p_stop, when in state S1 generate an error with probability p_error - would also work for the use case I have in mind (testing error correcting codes against the type of burst error that is introduced by a Viterbi decoder).

If people would prefer the simpler Markov model, I can implement it.

michael-roe, 2022-08-04:
markov_source implementation, documentation and tests updated to a new (and simpiler) specification.
Does that look any better?