gnuradio/gnuradio #6025: 4.0 build fails with Meson 0.63

haakov, 2022-07-25:
**What happened?**
`meson setup build` fails with the following error: `subprojects/cppzmq/meson.build:456:0: ERROR: Sandbox violation: Tried to grab file catch_with_main.cpp outside current (sub)project.`

The error disappears after downgrading Meson from 0.63 to 0.62.2.

**OS:** Arch Linux
**GR Installation Method:** Source

**GNU Radio Version**
4.0 (dev-4.0)

**Steps to Reproduce the Problem**
Try to build the dev-4.0 branch with Meson 0.63.

**Relevant log output**

`Executing subproject cppzmq method cmake

cppzmq| Found CMake: /usr/bin/cmake (3.23.2)

| Configuring the build directory with CMake version 3.23.2
| Running CMake with: -G Ninja -DCMAKE_INSTALL_PREFIX=/usr/local
|   - build directory:          /home/hkon/gr4/build/subprojects/cppzmq/__CMake_build
|   - source directory:         /home/hkon/gr4/subprojects/cppzmq
|   - toolchain file:           /home/hkon/gr4/build/subprojects/cppzmq/__CMake_build/CMakeMesonToolchainFile.cmake
|   - preload file:             /usr/lib/python3.10/site-packages/mesonbuild/cmake/data/preload.cmake
|   - trace args:               --trace-expand --trace-format=json-v1 --no-warn-unused-cli --trace-redirect=cmake_trace.txt
|   - disabled policy warnings: [CMP0025, CMP0047, CMP0056, CMP0060, CMP0065, CMP0066, CMP0067, CMP0082, CMP0089, CMP0102]

| Running with expanded trace output on.
| Not searching for unused variables given on the command line.
| Trace will be written to cmake_trace.txt
| -- Detected CPPZMQ Version - 4.8.1
| -- The C compiler identification is GNU 12.1.0
| -- The CXX compiler identification is GNU 12.1.0
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Check for working C compiler: /usr/bin/cc - skipped
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Check for working CXX compiler: /usr/bin/c++ - skipped
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| -- CMake libzmq package not found, trying again with pkg-config (normal install of zeromq)
| -- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.0")
| -- Found ZeroMQ: /usr/lib/libzmq.so
| -- cppzmq v4.8.1
| -- Looking for pthread.h
| -- Looking for pthread.h - found
| -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
| -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
| -- Found Threads: TRUE
| -- Configuring done
| -- Generating done
| -- Build files have been written to: /home/hkon/gr4/build/subprojects/cppzmq/__CMake_build

cppzmq| CMake configuration: SUCCEEDED
cppzmq| CMake project cppzmq has 33 build targets.

cmake-ast| Processing generated meson AST
cmake-ast| Build file: /home/hkon/gr4/build/subprojects/cppzmq/meson.build

cppzmq| Project name: cppzmq
cppzmq| Project version: undefined
cppzmq| C++ compiler for the host machine: c++ (gcc 12.1.0 "c++ (GCC) 12.1.0")
cppzmq| C++ linker for the host machine: c++ ld.bfd 2.38

subprojects/cppzmq/meson.build:456:0: ERROR: Sandbox violation: Tried to grab file catch_with_main.cpp outside current (sub)project.`

mormj, 2022-07-25:
@haakov - I was running into this prior to the following line being added:
`opt_var.add_cmake_defines({'CPPZMQ_BUILD_TESTS': false})`
https://github.com/gnuradio/gnuradio/blob/406f37e1a0664c87eb8828a1d21b0068bb1ba129/meson.build#L52

catch seems to be a dependency of the cppzmq tests, so when it doesn't build the tests it shouldn't hit this error.

I think the preferred solution is to just have cppzmq be a packaged dependency, but there are 2 problems with this right now:
1) the dependency is named differently between fedora and ubuntu (libzmq vs cppzmq)
2) the packaged version have different API changes requiring the #ifdefs in the code, which I was trying to avoid.  This is easily solvable by copying all the #ifdefs from GR3 ZMQ library

marcusmueller, 2024-01-31:
@mormj is meson-build still a feature? If yes, remove "clarification needed" and maybe add a meson and bug tag?

mormj, 2024-01-31:
No, the `dev-4.0` branch is dead - so this can be closed

closed 2024-01-31 by mormj