gnuradio/gnuradio #6026: dev-4.0 ninja build step fails

dl1ksv, 2022-07-25:
OS Linux, F35

**What happens**

Building dev-4.0 from source fails with

[1/357] Generating blocklib/analog/sig_source/gen_sig_source_srcs with a custom command
FAILED: blocklib/analog/sig_source/sig_source.cc blocklib/analog/sig_source/sig_source.h blocklib/analog/sig_source/sig_source_pybind.cc blocklib/analog/sig_source/analog_sig_source.block.yml 
/usr/bin/python3 /home/schroer/gnuradiocomponents/gnuradio-4.0/utils/blockbuilder/scripts/process_folder.py --yaml_file ../blocklib/analog/sig_source/sig_source.yml --output-cc blocklib/analog/sig_source/sig_source.cc --output-h blocklib/analog/sig_source/sig_source.h --output-pybind blocklib/analog/sig_source/sig_source_pybind.cc --output-grc blocklib/analog/sig_source/analog_sig_source.block.yml --grc-index 0 --build_dir /home/schroer/gnuradiocomponents/gnuradio-4.0/build
Traceback (most recent call last):
  File "/home/schroer/gnuradiocomponents/gnuradio-4.0/utils/blockbuilder/scripts/process_folder.py", line 9, in <module>
    import jsonschema_default
ModuleNotFoundError: No module named 'jsonschema_default'

mormj, 2022-07-25:
Try `pip install jsonschema_default`

This is a dependency that I'd like to get away from but it is filling a gap for now to populate default values from the schema

dl1ksv, 2022-07-26:
This did it. Build no succeeds

closed 2022-07-26 by dl1ksv