gnuradio/gnuradio #6029: 4.0: Replacement for gr-qtgui

mormj, 2022-07-25:
### Feature Description

tl;dr - we need something better than the current qtgui to release with 4.0

There have been many requests for updates to the QT GUI Widgets (https://wiki.gnuradio.org/index.php?title=GSoCIdeas#QT_Widgets_Improvements)

We have options for graphical widgets that might be easier to implement or more powerful than the current set of QT widgets.  
1) There was a partial porting of some of the qtgui functionality under newsched - it got ugly
2) A start of pyqtgraph based widgets (https://github.com/mormj/gr4-pyqtgraph)
  a) These are nice and there is a good docking manager
3) CyberEther - could these blocks be made native for 4.0 (https://github.com/luigifcruz/CyberEther)

### Feature Urgency

high (impacts broad portion of ecosystem)

RalphSteinhagen, 2022-07-26:
Food for thought: what about investing in [ImPlot](https://github.com/epezent/implot) based on the popular [Dear ImGui](https://github.com/ocornut/imgui) framework?  

It's quite lean, cross-platforms, and supports many backends such as Qt, SDL, Vulkan, Metal, etc.

In terms of programming style, it follows a 'game loop' style of programming that is also common in real-time signal processing.

Also, there is a [ImPlot demo](https://github.com/pthom/implot_demo) that transpiles nicely into WebAssembly if there is ever the need to send to and render GR data in web-applications, see the corresponding [web example](https://traineq.org/implot_demo/src/implot_demo.html) for illustration:

[image]

mormj, 2022-07-26:
@RalphSteinhagen - ImPlot looks really nice, and could be a very interesting way to modernize.  Thanks for the suggestion!

mcdiarmid, 2022-07-27:
Quick thought that doesn't directly assist in picking a GUI framework to ship with 4.0.  Thinking out loud here, no offence taken if this is flat-out rejected.  Would it be worth putting development effort into rethinking how autogenerated GUI code and `gr.top_block` interface?  More specifically, decoupling all of the GUI stuff from the options block, and having your Qt (or other framework) GUI Application being defined in another block.

Looking at python files generated by a flowgraph with GUI options "Qt GUI" and "No GUI", the only difference I see is Qt boilerplate code in the imports section, top block __init__, and in the main function.  I'm probably grossly oversimplifying this as I'm not familiar at all with the codebase used to generate these files.  But would it perhaps make more sense to have a Qt GUI "Application" or "Main Widget" block that is inherited by each of the GUI blocks?  All of the boilerplate could be handled in that one block and you end up with a the generated python file being a bit more consistent regardless of GUI framework, or lack thereof.  Could also take this a step further and make all other GUI widget blocks framework independent, all inheriting from the some abstract base class.  If this GUI module has an implementation for the widget block, create it, if not (or GUI framework is None/No GUI), the block defaults to a null sink/source/pass-through.

From my perspective this would also make out-of-tree GUI module development a lot friendlier.

RalphSteinhagen, 2022-07-27:
Another Idea may be to decouple GR's signal processing and UI aspects entirely using an intermediate interface layer that would expose the data via standard C++ stl containers which in turn would also allow splitting the two aspects into different threads or even processes.

The usual suspects for immediate interfaces are 
  1. lock-free circular buffer, e.g. Disruptor pattern, allowing fastest in-process data exchange, 
  2. a ZeroMQ PUB/SUB socket pair for fast in-process, across-process, across-host communication, or even
  3. a REST endpoint that would allow cross-platform including GR<->Web communications.
  
And the end-user/UI data interface being something like:
```cpp
std::list<std::pair<std::string, std::data<float>>>             signalData;     // signal < name, data>
std::list<std::pair<std::string, std::unordered_map</* .. */>>> signalMetaData; // tags, PMTs, etc.
// [..]
```  

For GRC this could be obviously bundled in the same application but this interface would permit also a plethora of external applications that do not necessarily need to compile against the GR libs and/or could avoid cross-lib interferences*.

Food for thought.

<sup>
*N.B. We started with such a decoupling scheme. One of the remaining (presently not critical) issues is a) how to synchronise the various independent UI/C++ sinks and have the work functions called synchronuously and b) the need for a singleton registry that contains the list of all declared UI-sinks (works but bit of an old-fashioned design).
</sub>

mormj, 2022-07-27:
@mcdiarmid - this is a very good and valid observation/suggestion.  The idea of GRC being more modular in this way has been proposed [here](https://github.com/gnuradio/greps/blob/main/grep-0025-grc-out-of-tree.md) with the idea of multiple "workflows" modularly defined for GRC.  I like the idea of this being a block that gets plunked down on the canvas though.

> From my perspective this would also make out-of-tree GUI module development a lot friendlier.

You are definitely right - there is not much room to improve GUI module development currently without adding a bunch of special cases to GRC

mormj, 2022-07-27:
@RalphSteinhagen - trying to get my head around what the implications here would be:
1) Is this pattern consistent with the GR buffer architecture?  Reading up on Disruptor pattern, it seems we have through the buffers and buffer readers multiple downstream consumers of a lock free queue.  This seems like it would be difficult to decouple from the current architecture
2) In this case would the result look like a "universal GUI sink" block that serializes to ZMQ, then GUI sinks can be anything that subscribes to that?

Just thinking out loud to try and piece this together - if GRC then rendered a non-GUI python/c++ script, but spewing data out of network interfaces, then the consumer script that launches the GUI could be rendered from a more GUI design friendly application

RalphSteinhagen, 2022-07-27:
@mormj no worries, hope my proposal isn't too bold.

Yes, this would be kind of a "universal sink" but not just limited to usage for GUIs. The lockfree multi-producer-multi-consumer isn't that difficult. We have an [implementation](https://github.com/fair-acc/opencmw-cpp/blob/main/src/disruptor/include/disruptor/RingBuffer.hpp) and [example](https://github.com/fair-acc/opencmw-cpp/blob/main/concepts/disruptor/QueueBenchmark.cpp) could sponsor if needed and nobody else has a better implementation.

We are updating our version of this "universal sink" as part of our OpenDigitizer project as an OOT module. Maybe, if you are interested and once it's finished/QA tested, you could have a look/review and we take it from there?
 ETA ~September (we are ⛺ing)

dkozel, 2022-07-28:
> decoupling all of the GUI stuff from the options block, and having your Qt (or other framework) GUI Application being defined in another block.

I strongly agree with decoupling all generation of GUI code from the core and continuing with our goal of having OOT and in-tree blocks be on equal tiers. OOT GUI options like gr-bokeh shouldn't require modifications in-tree.

I think pulling GRC out as a separate project is a separate issue from the handling of the GUI code and libraries used by applications using GNU Radio. I'd encourage folks not to try addressing both GRC design (single purpose utility application) with GNU Radio end user GUI interface design (Flexible, reconfigurable, possibly high performance demands, varying levels of customization). GRC's GTK or Qt interface and the gr-qtgui/gr-bokehgui modules are *very* separate topics. @mormj I'd like to recommend dividing [GREP 25](https://github.com/gnuradio/greps/blob/main/grep-0025-grc-out-of-tree.md) into a GUI module isolation GREP and one to convert GRC into a separate repo, which we can include as a submodule in the main GNU Radio repo if we choose.

## ImGUI

The CyberEther plots that Luigi has ported to GNU Radio use ImGUI
https://github.com/luigifcruz/gr-cyber

As does SDR++ by Alexandre Rouma.
https://github.com/AlexandreRouma/SDRPlusPlus

Anyone interested in starting a gr-imgui OOT as a demonstrator? If that and Josh's gr4-pyqtgraph are both able to integrate with the GNU Radio 4.0 GRC then at least the ability to swap application GUI options will be proven out.

## Intermediate interface layer for GUIs

I don't understand your idea @RalphSteinhagen. At least in part that sounds like what already exists? Can you clarify that a bit?

GNU Radio currently provides a buffer to GUI implementations which then process the data and choose how to display it, possibly working with tags or messages along side it. The GUI can then emit messages or write into buffers to pass signal and control information back to the DSP. Absolutely there are improvements possible, but is this at least similar to what you're describing?

Additionally there are a decent number of applications using GNU Radio for the DSP and then using the ZMQ or networking Sinks to pass the data out to a separate process which is displaying the UI. With 4.0 there has been discussion of formalizing a native interface for inter-process control/metadata passing.

Without totally understanding the proposal, this also sounds like it could either be a separate GUI module, "gr-guiinterface", or could be built into the core and the gr-qtgui, pyqtgraph, imgui, and bokehgui modules rewritten to use this new interface layer. I think you are imagining the later, but is the gr-guiinterface idea plausible?

RalphSteinhagen, 2022-07-30:
@dkozel no worries and thanks for the additional bits of info.

> I don't understand your idea @RalphSteinhagen. At least in part that sounds like what already exists? Can you clarify that a bit?

Yes, this could (and for us it will) totally be a separate OOT module. At the same time it could be reused to simplify, decouple, and enhance the compositability for UI and non-UI alike.

At the core: the present block API relies largely on inheritance, GR specific types, custom build system and rather large number of third party dependencies that may cause incompatibility issues (e.g. in the past some boost libraries, lately (unfortunately) also apparently {fmt} if not used as header-only lib). These are explicitly/implicitly pulled in when combining GR with other projects and eco systems. Inheritance-based API definition make it also harder to exchange API definitions and implementation.

In a nutshell what I was proposing:: add a generic sink that encapsulates/translate the internal GR structures and types to a generic  functional programming style interface that relies _only_ on STL container (or even none by using templates/concepts <-> C++20**) and only those 3rd party libs that are essential to the block functionality (even better: only header-only libs).

This would make it easier to extend, exchange fuctionality, and perhaps improve long-term stability of the API and also for internal/external UIs.

I wasn't aware of the GREP you mentioned, sorry.
Whether this affects/extends to the GRC design and should be used there is maybe another issue to discuss. Sorry again, I wasn't aware of the existing baggage if any. For what it's worth ImGUI brings only a few and -- depending on the chosen backend -- rather harmless third party dependencies along.

<sup>**P.S. re C++20: this would help the design from a C++ point of view but I am unsure how or if  Python and pybind11 can handle this.**</sup>

mormj, 2022-10-05:
I'm starting to get familiarized with imgui/implot and get my head around it - looks like a very valid path.  For any plot, as you mentioned we would just need a generic sink block with circular buffer (or being really tricky, just using the output buffer of the previous block) - grab the latest data either via callback/zmq/REST (maybe gated by an update rate) and update in the gui event loop.  Which is very similar to what gr-cyber is doing, but pulled up a level.

Combined with an OOT GRC with modular workflows, this could be a nice option.

dcajacob, 2022-10-06:
Might be out of scope, but I've always been frustrated when building a GUI in GRC, having to faff around getting the widgets where I want them. It seems like we could have a GUI Builder where you drop the GUI elements onto a canvas and move them around via point and click, resize, etc. Get it where you want it and then run it.