gnuradio/gnuradio #6032: 4.0: Remove jsonschema_default dependency

mormj, 2022-07-26:
### What happened?

This dependency is easily obtainable through pip, but not something that we should be relying on.  It was filling a gap to be able to generate a default block description and update specified parameters.  But this could be done without the dependency

### System Information

Source

### GNU Radio Version

3.11-git (main)

### Specific Version

4.0.0.0preview0git

### Steps to Reproduce the Problem

Not really

AnsahMohammad, 2025-02-09:
Hey, i checked the codebase and couldn't find any `jsonschema_default` being used. Has it already been fixed ?

mormj, 2025-02-10:
This was for the original dev-4.0 codebase, so it has become irrelevant since that migrated to gnuradio/gnuradio4 forking the project at fair-acc/gnuradio4

closed 2025-02-10 by mormj