gnuradio/gnuradio #6035: 4.0: Move to C++20

mormj, 2022-07-26:
### Feature Description

By the time 4.0 comes out, all the major distributions should have solid C++20 support.

One particular feature that might be of use is atomic_flags - this might cut down on the mutex waiting time in the concurrent queue in the scheduler.

https://modernescpp.com/index.php/performancecomparison-of-condition-variables-and-atomics-in-c-20

### Feature Urgency

medium (would be nice to have in the near future)

marcusmueller, 2022-08-04:
Oh I'd love to think we'd have `std::span`; that's really just a thin and cheap wrapper around a contiguous piece of memory, but it would be super handy to pass around parts of e.g. a ring buffer, or a vector. We could extend e.g. `block_work_input` like

```C++
class block_work_input
{
// …
public:
// …
std::span<T> data()
{
    return {static_cast<T>(buffer->read_ptr()), n_items};
}
}
```
so that `for(auto sample : work_io.inputs()[0].data()) {/*…*/}` works.

Then, the whole std::ranges / std::view thing, where I can lazily define the operations to be done on some input range (like a vector, span, initializer list), and then only when I access the result it is computed:

```C++
namespace views = std::views;
auto squared_positives =
    (std::span(work_io.inputs()[0].items(), work_io.inputs()[0].n_items) |  // pipes - such UNIX, much wow
     views::filter([](auto v) { return v > 0; }) |   // filter first
     views::transform([](auto v) { return v * v; })  // square after
    );
```

marcusmueller, 2022-08-05:
compilation with std=c++20 fails – in logging. I know whom to beat up over that! (in other words, I'll fix that.)

mormj, 2022-08-18:
Fixed with #6050 - thanks @marcusmueller

closed 2022-08-18 by mormj