gnuradio/gnuradio #6045: Cannot create hier block with variable vector length

nils-werner, 2022-07-27:
### What happened?

When trying to create a hier block with a variable vector length parameter, the resulting YAML file has a static `vlen` field in its inputs and outputs.

The problem seems to be that [`port.vlen` in the generator](https://github.com/gnuradio/gnuradio/blob/main/grc/core/generator/hier_block.py#L102)

```python
p['vlen'] = var_or_value(port.vlen)
```

has the value `1` and not the name of the parameter `vlen` that I'm using. `var_or_value()` then has no effect.

### System Information

OS: Arch Linux
GR Installation Method: Conda

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.3.0

### Steps to Reproduce the Problem

Create this flowgraph

```yaml
options:
  parameters:
    author: ''
    catch_exceptions: 'True'
    category: '[GRC Hier Blocks]'
    cmake_opt: ''
    comment: ''
    copyright: ''
    description: ''
    gen_cmake: 'On'
    gen_linking: dynamic
    generate_options: hb
    hier_block_src_path: '.:'
    id: test
    max_nouts: '0'
    output_language: python
    placement: (0,0)
    qt_qss_theme: ''
    realtime_scheduling: ''
    run: 'True'
    run_command: '{python} -u {filename}'
    run_options: prompt
    sizing_mode: fixed
    thread_safe_setters: ''
    title: Not titled yet
    window_size: (1000,1000)
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [8, 8]
    rotation: 0
    state: enabled

blocks:
- name: pad_sink_0
  id: pad_sink
  parameters:
    affinity: ''
    alias: ''
    comment: ''
    label: out
    num_streams: '1'
    optional: 'False'
    type: complex
    vlen: vlen
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [200, 124.0]
    rotation: 0
    state: true
- name: pad_source_0
  id: pad_source
  parameters:
    affinity: ''
    alias: ''
    comment: ''
    label: in
    maxoutbuf: '0'
    minoutbuf: '0'
    num_streams: '1'
    optional: 'False'
    type: complex
    vlen: vlen
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [40, 124.0]
    rotation: 0
    state: true
- name: vlen
  id: parameter
  parameters:
    alias: ''
    comment: ''
    hide: none
    label: Vlen
    short_id: ''
    type: intx
    value: '1'
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [184, 8.0]
    rotation: 0
    state: true

connections:
- [pad_source_0, '0', pad_sink_0, '0']

metadata:
  file_format: 1
  grc_version: 3.10.3.0
```

and generate the YAML and Python block files. The output will be `vlen: 1` instead of `vlen: ${ vlen }`

willcode, 2022-09-13:
Another report here (3.8): https://lists.gnu.org/archive/html/discuss-gnuradio/2022-09/msg00027.html

graceyeung68, 2022-09-15:
Yes, I am experiencing the same problem as indicated in the above [comment](https://github.com/gnuradio/gnuradio/issues/6045#issuecomment-1245284637). Do we anticipate this bug to be fixed or we should find a work around? Thanks!

willcode, 2022-09-16:
I don't know of anything in the works at the moment. You could edit the auto-generated Python for now.

graceyeung68, 2022-09-16:
Looks like the warning described [here](https://wiki.gnuradio.org/index.php?title=Python_Block_with_Vectors) is a similar issue.

[image]
[image]

willcode, 2022-09-16:
Ah, I thought you were talking about a heir block. The embedded python block definitely has some limitations, and I'm not sure there's a way to fix this without completely changing how embedded blocks are managed by GRC.

graceyeung68, 2022-09-16:
I am talking about a hier block but just thought the error mentioned for this modified embedded python block for vectors is also what I am getting. So in other words it seems the hier block generated by GRC does not accommodate for variable input vector lengths.

willcode, 2022-09-16:
Same symptom, I guess. But a heir block should work right, I think. Embedded blocks are a special hack.

graceyeung68, 2022-09-16:
It looks like the hier block python code also has that io signature set to the default parameter value upon initialization without checking to see if there is a global parameter of the same name. So GRC is seeing that and complaining about the IO size mismatch.

willcode, 2022-09-16:
It almost looks like everything works internally, but GRC doesn't know how to deal with pads whose signature can vary. It doesn't know that setting v=2 as a param means that it should evaluate the vlen of the pads if they are set to v inside the hier block. So GRC would need to evaluate the vlen of hier block pads whenever a variable in the top level flowgraph changes.

Short story: hier blocks will probably be stuck with static pad typing for a while.

SumiSpam, 2023-08-22:
Just comment that variable vector length works on GNU Radio 3.7.13.5. (see .xml generated files). I'm stuck on that version because I have more than a hundred of small hier blocks using this feature.
I'm too lazy to manually edit all those yaml files every time I make a change, so I continue waiting.
Thank's for all your work!

willcode, 2023-08-22:
Note: related to #4196

marcusmueller, 2024-02-01:
@nils-werner I *think* this issue is fixed now; when I run your .grc through `grc -o . repro6045.grc`, I do get

```python
# -*- coding: utf-8 -*-

#
# SPDX-License-Identifier: GPL-3.0
#
# GNU Radio Python Flow Graph
# Title: Not titled yet
# GNU Radio version: v3.11.0.0git-637-g26d8c762

from gnuradio import gr
from gnuradio.filter import firdes
from gnuradio.fft import window
import sys
import signal

class test(gr.hier_block2):
    def __init__(self, vlen=1):
        gr.hier_block2.__init__(
            self, "Not titled yet",
                gr.io_signature(1, 1, gr.sizeof_gr_complex*vlen),
                gr.io_signature(1, 1, gr.sizeof_gr_complex*vlen),
        )

        ##################################################
        # Parameters
        ##################################################
        self.vlen = vlen

        ##################################################
        # Connections
        ##################################################
        self.connect((self, 0), (self, 0))

    def get_vlen(self):
        return self.vlen

    def set_vlen(self, vlen):
        self.vlen = vlen

```

which would seem to be correct. Would you agree?

SumiSpam, 2024-03-14:
I have installed 3.10.9.2 version of GNU Radio on Windows 10 through Radioconda, and this issue remains unsolved.

The problem persist in the generated .yml file.

Regards!

[image]