gnuradio/gnuradio #6058: Trellis Metrics

michael-roe, 2022-08-08:
### What happened?

Using gnuradio-companion, I added a Trellis Metrics block with parameters output cardinality: 4, dimensionality: 1, constellation (1, 3, -1, -3), metric type Euclidean. When I ran it, I got the following error message show below.

### System Information

OS: Linux, Ubuntu 22.04
GR Installation Method: built from source from the github repo

### GNU Radio Version

3.11-git (main)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

-

### Relevant log output

```shell
Traceback (most recent call last):
  File "/home/mr101/src/gnuradio/test_trellis.py", line 192, in <module>
    main()
  File "/home/mr101/src/gnuradio/test_trellis.py", line 170, in main
    tb = top_block_cls()
  File "/home/mr101/src/gnuradio/test_trellis.py", line 83, in __init__
    self.trellis_metrics_x_0 = trellis.metrics_c(4, 1, (1, 3, -1, -3), digital.TRELLIS_EUCLIDEAN)
AttributeError: module 'gnuradio.trellis' has no attribute 'metrics_c'
```

n1ai, 2022-08-09:
I'm pretty confident the following issue has the fix you need:

https://github.com/gnuradio/gnuradio/pull/6022

Commit is:

https://github.com/gnuradio/gnuradio/pull/6022/commits/f76ebef7f1c73f3fc3df48524df3bf7f98acdbbe

Is in the main branch.

See also:

https://github.com/gnuradio/gnuradio/issues/5985

michael-roe, 2022-08-10:
Yes, that sounds like the same bug. I'll git pull the latest source, recompile, and see if it fixes the problem.

michael-roe, 2022-08-10:
Yes, this bug is a duplicate of #5985 and has been fixed.

closed 2022-08-10 by michael-roe