gnuradio/gnuradio #6066: GR 3.10.3 installing python code in wrong directory

cinaeds, 2022-08-09:
### What happened?

GR 3.10.3 installing python code in wrong directory. 

The installation path was set to /opt/gnuradio but the python code only was installed in /opt/lib/gnuradio - the binraries were installed in the correct directories

Moving the python code from /opt/lib to /opt/gnuradio/lib fixed the problem. 

This happens for all the OOT modules which generate python code. 

### System Information

OS: Debian bookworm  ARM64
GR Installation Method: Source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.3.0

### Steps to Reproduce the Problem

Installed volk using a recursive clone and using branch main (meant to use maint.) For got to do a git submodule update --init.

Use source code for maint-3.10 banch for gnuradio. Install in /opt/gnuradio.

Look in /opt/lib for the python code. 

Did a git submodule update for volk. 

All OOT modules with python code install in /opt/lib instead of /opt/gnuradio/lib.

I just move the python code from /opt/lib to /opt/gnuradio/lib and everything is honky dory.

### Relevant log output

```shell
I can capture the output of the build for gnuradio if needed.
```

ryanvolz, 2022-08-10:
Try updating so that you have [this recent commit](https://github.com/gnuradio/gnuradio/pull/6064) and see if that fixes it.

cinaeds, 2022-08-11:
Hi Ryan - it worked! Thank you!

-- Cinaed

closed 2022-08-19 by mormj