gnuradio/gnuradio #6074: DQPSK constellation object is not correct

marcinsztajn, 2022-08-10:
### What happened?

Using Constellation Object block there is an option to choose DQPSK constellation. In general, there is no such a thing like DQPSK constellation so the idea is to delete this class or replace with new constellation like 64 QAM. 

### System Information

OS: Linux Distro
GR Installation Method: apt

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.3.0

### Steps to Reproduce the Problem

Not related

### Relevant log output

```shell
Not related
```

jdemel, 2022-08-11:
Since we don't know if and how people use the DQPSK constellation object, we may deprecate it but we can't delete it. A new constellation like 64QAM would be a nice addition but it'd be unrelated to this issue.

The questions I'd like to discuss are:
- What's broken?
- How can one reproduce this issue?
- What should actually happen?
- Which are questions arise?

@marcinsztajn I assume that was you too, right?

> I did a simple tests and I think that constellation mapping has nothing to do with differential encoding. What I mean is you can rearrange your constellation points in whatever manner you would like but D - in front of DQPSK modulation scheme says that bytes before mapper are differentially encoded. Thus I think that [here](https://github.com/gnuradio/gnuradio/blob/bdd49943b6eda1004d5bf20e294639ee047ab40a/gr-digital/lib/constellation.cc#L718) is a mistake cause it shouldn't exist such a thing like DQPSK constellation object.

For reference [the section on differential modulation in PySDR](https://pysdr.org/content/digital_modulation.html#differential-coding).

I took the liberty to assign @777arc to this issue as well.

marcinsztajn, 2022-08-11:
So answering to @jdemel questions:

> What's broken?

Actually nothing is broken but the conception here is wrong. In my opinion terminology and the idea is wrong here. What I mean is that modulation scheme is missinterpreted as Constellation Object. To leverage the differential nature of DQPSK modulation scheme, the process of changing bit to symbols should involve (let me use GNU Radio terminology) Differential Encoder block followed by Chunks To Symbol block. The Chunks To Symbol block parameter called Symbol Table should use the constellation provided by given Constellation Object (like BPSK, QPSK, 16QAM etc.). That's how I understand the differential modulation process. I have created a simple flowgrah and linked it to the [Differential Encoder](https://wiki.gnuradio.org/index.php?title=Differential_Encoder) wiki - might help here. Please, verify my understanding.

> How can one reproduce this issue?

You can reproduce this by creating the flowgraph with one bitstream source followed by two branches: 
- connected to Chunks to Symbol block laveraging existing DQPSK Constellation Object symbol mapping,
- connected to Differential Encoder followed by Chunks To Symbol block which uses QPSK Constellation Object symbol mapping.
The output symbols would differ from each other.

> What should actually happen?

In my opinion DQPSK object is a wrong concept.

> Which are questions arise?

Question is, does my explaination above is correct in term of Comms theory.

jdemel, 2022-08-16:
> You can reproduce this by creating the flowgraph with one bitstream source followed by two branches:
>
> * connected to Chunks to Symbol block laveraging existing DQPSK Constellation Object symbol [...]

Since these are different modulations, I would expect a different outcome. A differential modulation scheme produces the next symbol depending on the previous one. I assume DBPSK is somehow misleading because it is possible to XOR bits and then just map them to BPSK symbols. This doesn't work as easily with other differential modulations.
[image]
I assume the above flowgraph is what you were refering to?

marcinsztajn, 2022-08-16:
> I assume the above flowgraph is what you were refering to?

Not really. Below is the flowgraphh I was refering to. Upper branch is how the GR user thinks that DQPSK modulation works now. Lower branch is how in my opinion DQPSK modulation should look like. Thats why I think that DQPSK constellation is misleading.
[image]

> A differential modulation scheme produces the next symbol depending on the previous one

Exactly, and DQPSK constellation object doesn't implement anything similar to that. 

> I assume DBPSK is somehow misleading because it is possible to XOR bits and then just map them to BPSK symbols. This doesn't work as easily with other differential modulations

What do you mean? DBPSK works as you describe, encode and map to BPSK constellation - thats all. What is misleading in that?

jdemel, 2022-08-24:
I suspect that the "DQPSK" constellation object is only valid in combination with another block. However, GR is missing an example on "How to use this feature". I wouldn't expect the flowgraph you posted. Intuitively, I'd combine the `Differential Encoder` block with the `DQPSK Constellation Object` in two consecutive blocks.

The discussion in https://github.com/777arc/textbook/issues/73 included a hint to [this resource](http://www.ece.iit.edu/~biitcomm/research/references/Other/Tutorials%20in%20Communications%20Engineering/TUTORIAL%202%20-%20Differential%20Encoding.pdf).
They make a clear distinction between "differential bit sequence encoding" and "differential phase encoding". DQPSK only makes sense in the latter case. While DBPSK might work in both contexts up to some point.

marcinsztajn, 2022-09-09:
> I suspect that the "DQPSK" constellation object is only valid in combination with another block

In my opinion it's valid only in combination with `Differential Encoder` block, do you have any doubts?
> I wouldn't expect the flowgraph you posted

May I know why?
> I wouldn't expect the flowgraph you posted. Intuitively, I'd combine the Differential Encoder block with the DQPSK Constellation Object in two consecutive blocks.

You can combine `Constallation Object` with `Differential Encoder` using `Chunks to Symbol` block. It's exacly what I did on the posted flowgraphs. Can you explain what is wrong with my flowgraph?
> The discussion in https://github.com/777arc/textbook/issues/73 included a hint to [this resource](http://www.ece.iit.edu/~biitcomm/research/references/Other/Tutorials%20in%20Communications%20Engineering/TUTORIAL%202%20-%20Differential%20Encoding.pdf).

What is the source of this document and who wrote it? At least let's use well known references. On contrary I propose to take a look at Proakis book [here](https://bayanbox.ir/view/9054313655403765775/Digital-Communications-5th-Edition.pdf) page 195. Quote from the book:

> The encoding is performed by a relatively simple logic circuit preceding the modulator

> They make a clear distinction between "differential bit sequence encoding" and "differential phase encoding". DQPSK only makes sense in the latter case. While DBPSK might work in both contexts up to some point

I don't agree. Why are you telling that it works with DBPSK but not for DQPSK? The algorithm is the same. The information is still encoded in phase changes not in symbol themselves. Take a scrap of paper and try to do math it as I did and prove yourself that it really works. Btw. first time I saw something like DEBPSK. I am keen on posting this on [dsp stack](https://dsp.stackexchange.com/) to have a discussion with wider audience and make this discussion close finally ;)

marcusmueller, 2024-02-01:
>  In general, there is no such a thing like DQPSK constellation 

Yes, there is! For example the DAB standard uses that.

closed 2024-02-01 by marcusmueller