gnuradio/gnuradio #6086: Is async crc16 the right endiannes?

michael-roe, 2022-08-13:
### What happened?

I put the following PDU into an Async CR16 block:
0x64 0x07 0x42 0x00

and got the following output:
0x64 0x07 0x42 0x00 0xdd 0xf9

But ... online CRC calculators for CRC16-CCITT-FALSE ( initial value 0xffff, xorout 0x0000) give a CRC of 0xf9 dd.

.. which is the opposite endianness. Is it the wrong way round?

======

I have a second issue with this block, which is that the CRC I really wanted was x16+x12+x5+1, initial value 0x, xorout 0xffff, and the block isn't configurable enough to match the diverse range of CRC variants we find in real radio protocols. (This CRC is used by Yaesu System Fusion).

### System Information

OS: Linux
GR Installation Method: Built from guthub source

### GNU Radio Version

3.11-git (main)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

-

michael-roe, 2022-08-13:
from crc16_async_bb_impl.cc, line 60:

memcpy((void*)(bytes_out.data() + pkt_len), &crc, 2);

That's an endianness bug, isn't it? (Computer running the code is probably a little endian x86 architecture machine, network endian is big endian).

michael-roe, 2022-08-13:
qa_crc16_async_bb will also need fixing if we change the block to fix endianness.

marcusmueller, 2024-01-31:
fixed as per the above PRs

closed 2024-01-31 by marcusmueller