gnuradio/gnuradio #6101: GRC Display issues on macOS

dxxx9, 2022-08-18:
### What happened?

This issue was reported before, but the proposed solutions didn't work for me. (see #5613, #5654, [#4814 (reply)])(https://github.com/gnuradio/gnuradio/pull/4814#issuecomment-925190019)

A flickering/not reacting screen sadly makes it impossible to work with the grc interface. (see image below)

Thankful for any help!

[image]

### System Information

OS: macOS Monterey 12.3
GR Installation Method: brew

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.3.0

### Steps to Reproduce the Problem

What I've done:
I updated (with homebrew) to from `3.9.3.0` to `3.10.3.0` and the flickering/black screen appears when I open a new flowgraph.

I had this issue before (with an older version of grc). As it is some time ago I'm not exactly sure how I fixed it but I can remeber adding
```
xterm_executable = /usr/X11R6/bin/xterm
```
to the `[grc]` in the config file.

dxxx9, 2022-08-19:
So after testing things the issue disappeared somehow. Might be that restarting the iTerm2 application fixed this.

I'll reopen this issue if I can provide detailed steps to reproduce that bug.

closed 2022-08-19 by dxxx9

dkozel, 2022-08-19:
@dxxx9 That would be great, understanding the cause and/or fix would be useful info to anyone looking to solve this more permanently.